jeremiah-c-leary/vhdl-style-guide #671: VSG hangs when processing particular file

imd1, 2021-11-09:
**Environment**
VSG 3.3.2, Centos 7.6.1810, if_002 remove brackets

**Describe the bug**
VSG hangs when processing the attached file. This is a third party file from somewhere, so I exclude it, but the fact that VSG hangs is worth investigating

[fixed_pkg_2008.zip](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/7504567/fixed_pkg_2008.zip)

jeremiah-c-leary, 2021-11-09:
The hanging is due to the multipool.  If there is an error in processing a file, the other workers in the pool do not quit.

The reason the file fails to process is due to lines like this:
```vhdl
378   function \?=\ (L, R : ufixed) return STD_ULOGIC;
```
My tokenizer splits up `\?=\` into `\`, `?=`, `\`.

jeremiah-c-leary, 2021-12-12:
Hey @imd1,

I have a final update for this issue on the issue-671 branch.  When you get a chance, could you check it out.

I will schedule this for release 3.5.0

Thanks,

--Jeremy

imd1, 2021-12-31:
I don't have time to check this, but I'm happy to accept that you've fixed the problem

jeremiah-c-leary, 2022-01-01:
k, I have merged the change into master.

closed 2022-01-01 by jeremiah-c-leary