jeremiah-c-leary/vhdl-style-guide #673: Another unfixable if statement

imd1, 2021-11-09:
**Environment**
VSG 3.3.2, Centos 7.6.1810, if_002 set to remove brackets

**Describe the bug**
Code snippet:
```
      IF (RISING_EDGE(m_axis_aclk_tb)) THEN
          IF (rst_in_tb = '0' AND stop_checks_out_tb /= zeros(top_num_ips-1 DOWNTO 0)) THEN
              all_stim_comps_stop_checks_int <= '1';
          ELSIF
             (stop_checks_out_tb = zeros(top_num_ips-1 DOWNTO 0)) THEN
              all_stim_comps_stop_checks_int <= '0';
          END IF;
      END IF;
```
* in check mode, if_002 fires on line 1 and 2, if_035 fires on line 4
* after fixing, code becomes...

```
      if RISING_EDGE(m_axis_aclk_tb) then
        if rst_in_tb = '0' AND start_ack_tb = ones(top_num_ips - 1 downto 0) then
          all_stim_comps_ack_int <= '1';
        elsif (start_ack_tb = zeros(top_num_ips - 1 downto 0)) then
          all_stim_comps_ack_int <= '0';
        end if;
      end if;
```
* and if_002 on (new) line 4 remains as an unfixable error

Comments
* some form of conflict between if_035 and if_002
* when I run VSG a second time on the new file, the if_002 violation is fixed, so it's like VSG needs two gos at this code!

jeremiah-c-leary, 2021-11-09:
There seems to be a never ending set of variations.  If_035 and if_002 run in the same phase and I would say if_002 runs before if_035.  I can make it run afterwards, but I think I need to update if_002 to handle anything after the `elsif` to the first parenthesis.
```vhdl
   elsif -- some comment
          -- some other comment, with blank line below

   (condition)
```

jeremiah-c-leary, 2022-02-16:
Hey @imd1,

I ran your example against the latest on master and it does not require two passes.  This the output with your example:
```vhdl
  1
  2 architecture rtl of fifo is
  3
  4 begin
  5
  6   proc_label : process is
  7   begin
  8
  9     if RISING_EDGE(m_axis_aclk_tb) then
 10       if rst_in_tb = '0' and stop_checks_out_tb /= zeros(top_num_ips - 1 downto 0) then
 11         all_stim_comps_stop_checks_int <= '1';
 12       elsif        stop_checks_out_tb = zeros(top_num_ips - 1 downto 0) then
 13         all_stim_comps_stop_checks_int <= '0';
 14       end if;
 15     end if;
 16
 17   end process proc_label;
 18
 19 end architecture rtl;
```
there is the odd spacing after the elsif though.  I think I need to update `if_005` to not require parenthesis.

--Jeremy

jeremiah-c-leary, 2022-02-16:
Hey @imd1,

I pushed a fix for this issue to the issue-673 branch.  When you get a chance could you check it out.  Otherwise I plan to merge this is by 06/23/2022.

Thanks,

--Jeremy

PR #744 (2022-02-20): Updates for issue #673
Fixing rule if_005 to remove whitespace if there is not a parenthesis.

  1)  Updated tests
  2)  Updated rule

Resolves #673
closed 2022-02-20 by jeremiah-c-leary