jeremiah-c-leary/vhdl-style-guide #674: Another example where VSG needs to run twice

imd1, 2021-11-10:
**Environment**
VSG 3.3.2, Centos 7.6.1810

**Describe the bug**
If you have a multiline signal assignment in a process...
```
        tb_s_axis_tdata(31 downto 0)                <= std_logic_vector(to_unsigned(tg_chan_num, 8)) & "0000000" & tg_daf_flag &
                                                       std_logic_vector(to_unsigned(tg_sample_num, 8)) & std_logic_vector(to_unsigned(1, 8));

```
the rule that aligns the `<=` runs and moves it accordingly, but this leaves the rule `sequential_004` reporting an unfixable error about trying to line up the second line of the assignment.

Another example of two rules sort of running in the wrong order

The workaround is to execute VSG again to fix the `sequential_004` violation

jeremiah-c-leary, 2022-01-31:
I have a fix for this with issue #680 so I will close this issue when I close the other issue.

--Jeremy

jeremiah-c-leary, 2022-02-05:
Hey @imd1,

I just merged this to master for inclusion in the next release.

--Jeremy

closed 2022-02-05 by jeremiah-c-leary