jeremiah-c-leary/vhdl-style-guide #675: Autofix missing process label

imd1, 2021-11-10:
**Is your feature request related to a problem? Please describe.**
Currently process_016 (missing process labels) requires a manual fix, presumably because VSG cannot hope to know what to call a process.

**Describe the solution you'd like**
However it would be useful for VSG to create a label using a predefined format, which would allow someone to search the file for this label and apply fixes, and would ensure the rule is not violated at least

For example:

```
VSG_AUTO_1: process is
..
VSG_AUTO_2: process is
```

The actual format could be specified in the config file?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jeremiah-c-leary, 2022-02-06:
Hey @imd1,

I could add an option for the label contents the user could modify:
```yaml
  rule:
     process_016:
         disable: False
         label: 'MY_PROC'
```
Then VSG could increment a number after the label:
```vhdl
   MY_PROC1 : process

   MY_PROC2 : process
```
The rule would be disabled by default, but the error would point the user to the rule documentation which would then show how to enable and autogenerate labels if they want.

What do you think?

--Jeremy

imd1, 2022-02-06:
Good idea. Also adding a //TODO comment would be useful because this shows up in modern IDEs

jeremiah-c-leary, 2022-02-13:
So something like this:

```vhdl
  MY_PROC : process (....)   -- TODO: Rename process label
```

I was just wondering if this really addresses the issue of process labels.  With this feature, the user still has to update the labels.  What is the user gaining with this feature?

--Jeremy

jeremiah-c-leary, 2022-10-17:
Hey @imd1 ,

I decided not to implement this feature so I will be closing this issue.  If you would like to discuss this further just re-open the issue and we can explore it.

Thanks,

--Jeremy

closed 2022-10-17 by jeremiah-c-leary