jeremiah-c-leary/vhdl-style-guide #676: User defined attributes case handling

imd1, 2021-11-10:
**Environment**
VSG 3.3.2, Centos 7.6.1810

**Describe the bug**
entity_specification_502 rules specifies the case of user defined attributes. However this rule can fire for attributes defined for signals, constants, components etc, and each of these has its own case rules. 

Problem I have is that I have conflicting case rules for constants etc and so cannot set this rule properly

I have to disable this rule as a workaround

Possible solution is to entend the scope of case rules for constants, signals etc to include application to user defined attributes

jeremiah-c-leary, 2021-12-04:
Maybe `entity_specification_502` should be removed as the case should be handled by a consistent case type rule.

What do you think?

--Jeremy

imd1, 2021-12-31:
Not sure how to proceed...somehow it makes sense to extend the scope of existing rules to cover labels in user defined attributes rather than define a new set of rules that must match the existing settings

jeremiah-c-leary, 2022-02-12:
Hey @imd1,

> somehow it makes sense to extend the scope of existing rules to cover labels in user defined attributes rather than define a new set of rules that must match the existing settings

I agree.  I will remove this rule.

--Jeremy

jeremiah-c-leary, 2022-02-12:
Hey @imd1,

I pushed a fix for this to the issue-676 branch.  I just removed the rule, so I could just merge it to trunk.

I plan to push this to master by 02/09/2022.

Thanks,

--Jeremy

PR #735 (2022-02-20): Updates for issue #676
Deprecating rule entity_specification_502.

The entity_designator should be enforced by a consistent case rule.

fixes #676
closed 2022-02-20 by jeremiah-c-leary