jeremiah-c-leary/vhdl-style-guide #677: Unfixable multiline constant

imd1, 2021-11-10:
**Environment**
VSG 3.3.2, Centos 7.6.1810

**Describe the bug**
The following code generates unfixable `constant_012` errors, even though the suggested fix reported is just re-enforcing the existing alignment (mostly)!

```
architecture rtl of an_entity is

  constant ROM : rom_table := rom_table'(
                                         ROM_WORD'(STD_LOGIC_VECTOR(TO_UNSIGNED( 1, 5))),
                                         ROM_WORD'(STD_LOGIC_VECTOR(TO_UNSIGNED( 0, 5))),
                                         ROM_WORD'(STD_LOGIC_VECTOR(TO_UNSIGNED( 1, 5))),
                                         );

begin

end architecture rtl;
```
```
================================================================================
File: /path/to/file.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 415
Total Violations:    4
  Error   :     4
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  constant_012              | Error      |          4 | Adjust indent to column 42
  constant_012              | Error      |          5 | Adjust indent to column 42
  constant_012              | Error      |          6 | Adjust indent to column 42
  constant_012              | Error      |          7 | Adjust indent to column 40
----------------------------+------------+------------+--------------------------------------

```
**Additional context**
Add any other context about the problem here.

jeremiah-c-leary, 2022-02-16:
Hey @imd1,

I just ran your example against the latest on master and it did not complain after fixing.  Although the closing parenthesis is off after the fix:
```vhdl
  1
  2 architecture rtl of an_entity is
  3
  4   constant rom : rom_table := rom_table'(
  5                                          ROM_WORD'(std_logic_vector(TO_UNSIGNED( 1, 5))),
  6                                          ROM_WORD'(std_logic_vector(TO_UNSIGNED( 0, 5))),
  7                                          ROM_WORD'(std_logic_vector(TO_UNSIGNED( 1, 5))),
  8                                        );
  9
 10 begin
 11
 12 end architecture rtl;
```
I will track that part down as part of this issue.

Could you validate on your end if you see the expected behavior?

Thanks,

--Jeremy

jeremiah-c-leary, 2022-02-16:
Actually, I would like to close this issue and include it in issue #699.  I will be reviewing all the multiline alignment rules and trying to consolidate them.

--Jeremy

jeremiah-c-leary, 2022-07-04:
Hey @imd1,

Just wanted to ping you about this issue.  I can not duplicate it.  If I do not hear back from you by July 15th, I will assume this is no longer an a problem and will close the issue.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-07-15:
Hey @imd1,

I am going to close this issue.  If you run into this again please let me know.

Regards,

--Jeremy

closed 2022-07-15 by jeremiah-c-leary