jeremiah-c-leary/vhdl-style-guide #678: Another unfixable multiline constant

imd1, 2021-11-11:
**Environment**
VSG 3.3.2, Centos 7.6.1810

**Describe the bug**
Original code:
```
architecture simulation of an_architecture is

  constant c_timing_slot : t_timing_slot(0 to 31) :=(
  B"0000_0000_0000_0000_0000_0000_0000_0000",
  B"0000_0000_0000_0000_0000_0000_0000_0000");

begin

end architecture simulation;
```

Results of a check only:
```
================================================================================
File:  /path/to/file.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 415
Total Violations:    4
  Error   :     4
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  constant_016              | Error      |          3 | Move parenthesis after assignment to the next line.
  constant_012              | Error      |          4 | Adjust indent to column 54
  constant_012              | Error      |          5 | Adjust indent to column 54
  constant_016              | Error      |          5 | Move closing parenthesis to the next line.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

Result after fix:
```
================================================================================
File:  /path/to/file.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 104
Total Violations:    2
  Error   :     2
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  constant_007              | Error      |          3 | move assignment to same line as constant declaration.
  whitespace_001            | Error      |          3 | Remove trailing whitespace
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

Fixed code:
```
architecture simulation of an_architecture is

  constant C_TIMING_SLOT : t_timing_slot(0 to 31) 
 :=(
     B"0000_0000_0000_0000_0000_0000_0000_0000",
     B"0000_0000_0000_0000_0000_0000_0000_0000"
   );

begin

end architecture simulation;
```

jeremiah-c-leary, 2022-02-06:
Hey @imd1,

I just pushed a fix for this to the issue-678 branch.  When you get a chance could you validate it works on your end?

I will plan to push this to master around 2/13/2022 if I do not hear back from you.

Thanks,

--Jeremy

imd1, 2022-02-06:
I cannot easily action your request so feel free to proceed

jeremiah-c-leary, 2022-02-07:
Will do, thanks,

--Jeremy

closed 2022-02-07 by jeremiah-c-leary