jeremiah-c-leary/vhdl-style-guide #679: Comment causes VSG crash

imd1, 2021-11-11:
**Environment**
VSG 3.3.2, Windows 10

**Describe the bug**
Attached file causes VSG to crash
[file.zip](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/7519434/file.zip)

A comment contains quotation marks that are ASCII 0x93 and 0x94

Resulting VSG output

```
Traceback (most recent call last):
  File "c:\users\imd\appdata\local\programs\python\python36\lib\multiprocessing\pool.py", line 119, in worker
    result = (True, func(*args, **kwds))
  File "c:\users\imd\appdata\local\programs\python\python36\lib\site-packages\vsg\apply_rules.py", line 83, in apply_rules
    write_vhdl_file(oVhdlFile)
  File "c:\users\imd\appdata\local\programs\python\python36\lib\site-packages\vsg\apply_rules.py", line 111, in write_vhdl_file
    oFile.write(sLine + '\n')
  File "c:\users\imd\appdata\local\programs\python\python36\lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode characters in position 6-7: character maps to <undefined>
```
**Additional context**
Add any other context about the problem here.

jeremiah-c-leary, 2021-12-04:
Hey @imd1,

That particular file did not crash for me.  Is that the minimal file that causes you an error?

--Jeremy

imd1, 2021-12-05:
> Hey @imd1,
> 
> That particular file did not crash for me. Is that the minimal file that causes you an error?
> 
> --Jeremy

Yes, note that running Python 3.6 under Windows, have not tried under any flavor of Unix

imd1, 2021-12-05:
Also note that this file requires fixes according to my rule setup...

```
================================================================================
File:  C:\Users\imd\Downloads\file\file.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 104
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  architecture_010          | Error      |          7 | Add *architecture* keyword.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

```

jeremiah-c-leary, 2021-12-07:
I downloaded python 3.6.8 to my windows machine and was able to replicate the error.  It looks like it could be a python thing:
```mono
File "c:\users\imd\appdata\local\programs\python\python36\lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode characters in position 6-7: character maps to <undefined>
```
since the encoding from the python lib is raising an error.

I am running with python 3.8.2 and do not see the issue.

Do you have access to python 3.7 to see if it shows the problem?

imd1, 2021-12-31:
> Do you have access to python 3.7 to see if it shows the problem?

Not easily, no

imd1, 2022-02-06:
Are we saying that we need a min Python version?

jeremiah-c-leary, 2022-02-07:
Possibly....Let me re-investigate first.

--Jeremy

eminakgun, 2022-02-08:
I had similar issues when file consist Turkish characters. I was running under Windows, Python 3.8.
Even though local encoding was recognized as cp1254(Turkish), see [python docs](https://docs.python.org/3/library/codecs.html#standard-encodings), Turkish characters were corrupted. When I set encoding explicitly as 'utf-8', issue resolves.

I applied the change in lines below,  
https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/master/vsg/vhdlFile/utils.py#L690
https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/master/vsg/apply_rules.py#L121

eminakgun, 2022-02-08:
#535

jeremiah-c-leary, 2022-02-10:
Hey @eminakgun,

Could you paste your changes to those files?

Thanks,

--Jeremy

eminakgun, 2022-02-11:
I will be able to do that when I'm on my windows machine.

jeremiah-c-leary, 2022-04-03:
Hey @eminakgun,

Just wanted to ping you for an example.

Thanks,

--Jeremy

eminakgun, 2022-04-04:
The changes I made back then were,

1. In [write_vhdl_file](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/master/vsg/apply_rules.py#L119) method
`with open(oVhdlFile.filename, 'w') as oFile:` was changed to `with open(oVhdlFile.filename, 'w', encoding='utf-8') as oFile:`
2. In [read_vhdlfile](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/master/vsg/vhdlFile/utils.py#L698) method
`with open(sFileName) as oFile:` was changed to `with open(sFileName, encoding='utf-8') as oFile:`

Without specifying encoding type, Python gets it from OS. In my case, even though it was correct, when file was read, characters were corrupted. AFAIK, utf-8 is capable of representing any Unicode character. So specifying utf-8 explicitly would'nt make any problem.

Example including Turkish characters.
```
entity uart_tx is
  generic (
    baudrate : integer := 115200;  -- şğçiö
    clk_f    : integer := 100000000 -- ÇŞĞÇİÖŞİ
  );
  port (
    clk          : in    std_logic; -- şğçiö
    send         : in    std_logic; -- şğçiö
    tx_dout      : out   std_logic; -- şğçiö
    tx_data_in   : in    std_logic_vector(7 downto 0); -- şğçiö
    tx_done_tick : out   std_logic -- şğçiö
  );
end entity uart_tx;
```

vsg output,
`PS C:\Users\emina\Desktop> vsg -f .\UART_TX.vhd --fix -b                                                                                 multiprocessing.pool.RemoteTraceback:                                                                                                    """                                                                                                                                      Traceback (most recent call last):                                                                                                         File "c:\python\python38\lib\multiprocessing\pool.py", line 125, in worker                                                                 result = (True, func(*args, **kwds))                                                                                                   File "c:\python\python38\lib\site-packages\vsg\apply_rules.py", line 95, in apply_rules                                                    write_vhdl_file(oVhdlFile)                                                                                                             File "c:\python\python38\lib\site-packages\vsg\apply_rules.py", line 123, in write_vhdl_file                                               oFile.write(sLine + '\n')                                                                                                              File "c:\python\python38\lib\encodings\cp1254.py", line 19, in encode                                                                      return codecs.charmap_encode(input,self.errors,encoding_table)[0]                                                                    UnicodeEncodeError: 'charmap' codec can't encode character '\x9f' in position 43: character maps to <undefined>                          """                                                                                                                                                                                                                                                                               The above exception was the direct cause of the following exception:                                                                                                                                                                                                              Traceback (most recent call last):                                                                                                         File "C:\Python\Python38\Scripts\vsg-script.py", line 33, in <module>                                                                      sys.exit(load_entry_point('vsg==3.9.0', 'console_scripts', 'vsg')())                                                                   File "c:\python\python38\lib\site-packages\vsg\__main__.py", line 138, in main                                                             for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):                                                                 File "c:\python\python38\lib\multiprocessing\pool.py", line 868, in next                                                                   raise value                                                                                                                          UnicodeEncodeError: 'charmap' codec can't encode character '\x9f' in position 43: character maps to <undefined>  `

P.S without upper case characters, vsg runs without crashing.

jeremiah-c-leary, 2022-04-11:
Hey @imd1 and @eminakgun,

I was finally able to validate the failure in Windows using powershell.  I added the changes @eminakgun suggested and it no longer crashed.  I wish I could have seen this crash under linux though.

I pushed a change for this up to the issue-679 branch.  When you get a chance could you try it out and let me know if it addresses the issue.

I will plan to merge this to master by 4/15 if I do not hear back.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-04-14:
Sweet, I will merge it in.

PR #753 (2022-04-14): Updates for issue #679
Fixing encoding issues when reading and writing files.

  1)  Added utf-8 encoding and decoding
  2)  Fixed issue with token map which was causing vsg to crash.

Resolves #679
closed 2022-04-14 by jeremiah-c-leary