jeremiah-c-leary/vhdl-style-guide #680: Yet another example where VSG must be run twice

imd1, 2021-11-11:
**Environment**
VSG 3.3.2, Centos 7.6.1810

**Describe the bug**

Original code
```
architecture simulation of an_entity is

BEGIN  
  
proc_a_process : process(tb_clk_sys)
begin
    if (tb_areset = '1') then
        frame_count <= 0;
    
    elsif rising_edge(tb_clk_sys) then
        
        tid_slv     := std_logic_vector(to_unsigned(tid, 8));
        data_word   <= std_logic_vector(to_unsigned(frame_count, 8)) &
                       std_logic_vector(to_unsigned(word_count, 8)) & 
                       std_logic_vector(to_unsigned(word_target(tid), 8)) & 
                       tid_slv(6 downto 3) & "0" & tid_slv(2 downto 0);
        data_tid    <= std_logic_vector(to_unsigned(tid, tb_s_axis_tid'length));
        data_valid  <= valid_flag;
 
    end if;
end process proc_a_process;

end simulation;

```

After first VSG run in fix mode:
```
architecture simulation of an_entity is

begin

  PROC_A_PROCESS : process (tb_clk_sys) is
  begin

    if tb_areset = '1' then
      frame_count <= 0;
    elsif rising_edge(tb_clk_sys) then
      tid_slv := std_logic_vector(to_unsigned(tid, 8));
      data_word  <= std_logic_vector(to_unsigned(frame_count, 8)) &
                     std_logic_vector(to_unsigned(word_count, 8)) &
                     std_logic_vector(to_unsigned(word_target(tid), 8)) &
                     tid_slv(6 downto 3) & "0" & tid_slv(2 downto 0);
      data_tid   <= std_logic_vector(to_unsigned(tid, tb_s_axis_tid'length));
      data_valid <= valid_flag;
    end if;

  end process PROC_A_PROCESS;

end architecture simulation;

================================================================================
File:  /path/to/file.vhd
================================================================================
Phase 4 of 7... Reporting
Total Rules Checked: 368
Total Violations:    3
  Error   :     3
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  sequential_004            | Error      |         13 | Adjust indent to column 20
  sequential_004            | Error      |         14 | Adjust indent to column 20
  sequential_004            | Error      |         15 | Adjust indent to column 20
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

After second run of VSG:
```
architecture simulation of an_entity is

begin

  PROC_A_PROCESS : process (tb_clk_sys) is
  begin

    if tb_areset = '1' then
      frame_count <= 0;
    elsif rising_edge(tb_clk_sys) then
      tid_slv := std_logic_vector(to_unsigned(tid, 8));
      data_word  <= std_logic_vector(to_unsigned(frame_count, 8)) &
                    std_logic_vector(to_unsigned(word_count, 8)) &
                    std_logic_vector(to_unsigned(word_target(tid), 8)) &
                    tid_slv(6 downto 3) & "0" & tid_slv(2 downto 0);
      data_tid   <= std_logic_vector(to_unsigned(tid, tb_s_axis_tid'length));
      data_valid <= valid_flag;
    end if;

  end process PROC_A_PROCESS;

end architecture simulation;

================================================================================
File:  C:\Users\imd\vhdl-lint\test/example/proj3/rfel_demux/test/rfel_demux_top_stress_tb.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 580
Total Violations:    0
  Error   :     0
  Warning :     0

0 violations

```

wespiard, 2022-01-08:
I run into this  a lot too with similar examples.

jeremiah-c-leary, 2022-01-31:
Hey @imd1 and @wespiard,

I just pushed an update for this issue to the issue-680 branch.  When you get a chance could you check it out and let me know if the update fixes the issue on your side?

I will plan to merge this into master after 02/07/2022 if I have not heard back from you.

Regards,

--Jeremy

wespiard, 2022-01-31:
I don't have any examples myself right now, but I will run @imd1's original example. Seems like the indentation error was resolved, but now there is an `if_002` error that remains.

```
architecture simulation of an_entity is

BEGIN  
  
proc_a_process : process(tb_clk_sys)
begin
    if (tb_areset = '1') then
        frame_count <= 0;
    
    elsif rising_edge(tb_clk_sys) then
        
        tid_slv     := std_logic_vector(to_unsigned(tid, 8));
        data_word   <= std_logic_vector(to_unsigned(frame_count, 8)) &
                       std_logic_vector(to_unsigned(word_count, 8)) & 
                       std_logic_vector(to_unsigned(word_target(tid), 8)) & 
                       tid_slv(6 downto 3) & "0" & tid_slv(2 downto 0);
        data_tid    <= std_logic_vector(to_unsigned(tid, tb_s_axis_tid'length));
        data_valid  <= valid_flag;
 
    end if;
end process proc_a_process;

end simulation;
```

```
❯ ~/git/vhdl-style-guide/bin/vsg --version
VHDL Style Guide (VSG) version: 3.6.0.dev2
Git commit SHA: g9a4fbfd3

❯ ~/git/vhdl-style-guide/bin/vsg -f test.vhd --fix
================================================================================
File:  test.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 106
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  if_002                    | Error      |         10 | Enclose condition in ()'s.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

And here is the resulting file after this first attempt: 

```
architecture simulation of an_entity is

begin

  proc_a_process : process (tb_clk_sys) is
  begin

    if (tb_areset = '1') then
      frame_count <= 0;
    elsif tb_clk_sys'event and tb_clk_sys = '1' then
      tid_slv    := std_logic_vector(to_unsigned(tid, 8));
      data_word  <= std_logic_vector(to_unsigned(frame_count, 8)) &
                    std_logic_vector(to_unsigned(word_count, 8)) &
                    std_logic_vector(to_unsigned(word_target(tid), 8)) &
                    tid_slv(6 downto 3) & "0" & tid_slv(2 downto 0);
      data_tid   <= std_logic_vector(to_unsigned(tid, tb_s_axis_tid'length));
      data_valid <= valid_flag;
    end if;

  end process proc_a_process;

end architecture simulation;
```

After running it again, it got rid of the remaining error: 

```
❯ ~/git/vhdl-style-guide/bin/vsg -f test.vhd --fix
================================================================================
File:  test.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 565
Total Violations:    0
  Error   :     0
  Warning :     0
```

And the final resulting file: 
```
architecture simulation of an_entity is

begin

  proc_a_process : process (tb_clk_sys) is
  begin

    if (tb_areset = '1') then
      frame_count <= 0;
    elsif (tb_clk_sys'event and tb_clk_sys = '1') then
      tid_slv    := std_logic_vector(to_unsigned(tid, 8));
      data_word  <= std_logic_vector(to_unsigned(frame_count, 8)) &
                    std_logic_vector(to_unsigned(word_count, 8)) &
                    std_logic_vector(to_unsigned(word_target(tid), 8)) &
                    tid_slv(6 downto 3) & "0" & tid_slv(2 downto 0);
      data_tid   <= std_logic_vector(to_unsigned(tid, tb_s_axis_tid'length));
      data_valid <= valid_flag;
    end if;

  end process proc_a_process;

end architecture simulation;
```

jeremiah-c-leary, 2022-01-31:
Thanks @wespiard,

I see the `if_002` error being asserted.

jeremiah-c-leary, 2022-01-31:
Another case of one rule running before another.  I pushed a fix for the `if_002` error.  If you could validate on your side I would appreciate it.

Thanks,

--Jeremy

wespiard, 2022-02-01:
@jeremiah-c-leary , on branch issue-680 I pulled, ran the python setup script, ran `vsg` the same as in my previous post and got the same error.

When I do `git log`, is says I'm on commit `36fbb21`.

jeremiah-c-leary, 2022-02-01:
Hey @wespiard,

That is odd, when I run it it doesn't complain about `if_002`.

Source file:
```vhdl

architecture simulation of an_entity is

BEGIN

proc_a_process : process(tb_clk_sys)
begin
    if (tb_areset = '1') then
        frame_count <= 0;

    elsif rising_edge(tb_clk_sys) then

        tid_slv     := std_logic_vector(to_unsigned(tid, 8));
        data_word   <= std_logic_vector(to_unsigned(frame_count, 8)) &
                       std_logic_vector(to_unsigned(word_count, 8)) &
                       std_logic_vector(to_unsigned(word_target(tid), 8)) &
                       tid_slv(6 downto 3) & "0" & tid_slv(2 downto 0);
        data_tid    <= std_logic_vector(to_unsigned(tid, tb_s_axis_tid'length));
        data_valid  <= valid_flag;

    end if;
end process proc_a_process;

end simulation;
```
Running this command:
```mono
$vsg -f original.vhd --fix
================================================================================
File:  fixed.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 565
Total Violations:    0
  Error   :     0
  Warning :     0
```
The resulting file is:
```vhdl

architecture simulation of an_entity is

begin

  proc_a_process : process (tb_clk_sys) is
  begin

    if (tb_areset = '1') then
      frame_count <= 0;
    elsif (tb_clk_sys'event and tb_clk_sys = '1') then
      tid_slv    := std_logic_vector(to_unsigned(tid, 8));
      data_word  <= std_logic_vector(to_unsigned(frame_count, 8)) &
                    std_logic_vector(to_unsigned(word_count, 8)) &
                    std_logic_vector(to_unsigned(word_target(tid), 8)) &
                    tid_slv(6 downto 3) & "0" & tid_slv(2 downto 0);
      data_tid   <= std_logic_vector(to_unsigned(tid, tb_s_axis_tid'length));
      data_valid <= valid_flag;
    end if;

  end process proc_a_process;

end architecture simulation;
```
Where you can see the elsif line was changed from `rising_edge` to an `event` and then enclosed with `()`.

This is what version is reporting:
```mono
$ vsg --version
VHDL Style Guide (VSG) version: 3.6.0.dev3
Git commit SHA: g36fbb211
```
Not sure where the `g` is coming from in the SHA though.
Here is the git log:
```mono
$ git log
commit 36fbb211312dbaea3fcdbfa4057d33a7cfd708cc (HEAD -> issue-680, origin/issue-680)
Author: Jeremiah Leary <jeremiah.c.leary@gmail.com>
Date:   Mon Jan 31 07:28:53 2022 -0600

    Fixed issue where process_029 was running after if_002 and removing the parenthesis.

      1)  Moved if_002 to subphase 2.

commit 9a4fbfd3530bbeb92c3a7331e58947c7d67b539e
Author: Jeremiah Leary <jeremiah.c.leary@gmail.com>
Date:   Mon Jan 31 04:45:01 2022 -0600

    Updates for issue #680

    Alignment of sequential assignments would take two passes to complete.

      1)  Isolated issue to process_004 being ran before process_400
      2)  Updated test
      3)  Set sequential_004 to run after process_400
```
and my git status:
```mono
$ git status
On branch issue-680
Your branch is up to date with 'origin/issue-680'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        issue_validation/
        release_notes.txt
        scripts/

nothing added to commit but untracked files present (use "git add" to track)
```
Could you double check your version maybe?  Not sure how to proceed.

--Jeremy

wespiard, 2022-02-01:
@jeremiah-c-leary My bad, just tested again and it works. 

My problem was that when I switched to the relevant branch/commit, then ran `python setup.py install`, it installed the new binary to a different location, so I was still running the non-fixed version. 

Now I did `pip uninstall vsg` to semove my old version, then ran `python setup.py install` and got the new version and it worked.

wespiard, 2022-02-01:
```
❯ vsg --version
VHDL Style Guide (VSG) version: 3.6.0.dev3
Git commit SHA: g36fbb211

~/tmp
❯ vsg -f test.vhd --fix
================================================================================
File:  test.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 565
Total Violations:    0
  Error   :     0
  Warning :     0
```

jeremiah-c-leary, 2022-02-02:
Hey @wespiard,,

Not a problem.  It usually ends up being either the wrong python version or the wrong vsg version.

jeremiah-c-leary, 2022-02-05:
Hey @wespiard and @imd1,

I just merged this to master for inclusion in the next release.

--Jeremy

closed 2022-02-05 by jeremiah-c-leary