jeremiah-c-leary/vhdl-style-guide #681: Unfixable whitespace

imd1, 2021-11-11:
**Environment**
VSG 3.3.2, Centos 7.6.1810

**Describe the bug**

Code:
```
architecture simulate of an_entity is

begin

  proc_a_process : process (all) is         -- tb_areset, tb_sys_clk
  begin

  end process proc_a_process;

end architecture simulate;
```

Unfixable error:
```
================================================================================
File:  /path/to/file.vhd
================================================================================
Phase 2 of 7... Reporting
Total Rules Checked: 206
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  whitespace_005            | Error      |          5 | Remove spaces after open (.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

```

imd1, 2022-02-06:
I cannot easily action your request so feel free to proceed

jeremiah-c-leary, 2022-02-07:
Will do, thanks,

--Jeremy

closed 2022-02-07 by jeremiah-c-leary