jeremiah-c-leary/vhdl-style-guide #682: More VSG double runs

imd1, 2021-11-11:
**Environment**
VSG 3.3.2, Centos 7.6.1810

**Describe the bug**
Original code:
```
entity adder_tree is
   generic ( in_width: integer := 14; out_width: integer := 15 ); 
   port (
         Q:       out STD_LOGIC_VECTOR( (out_width-1) downto 0 )
   );
end entity;
```

After first VSG fix:
```
entity adder_tree is
  generic (
    IN_WIDTH  : integer := 14;
    OUT_WIDTH : integer := 15 
  );
  port (
    Q : out   std_logic_vector((OUT_WIDTH - 1) downto 0)
  );
end entity adder_tree;

================================================================================
File:  /path/to/file.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 103
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  whitespace_001            | Error      |          4 | Remove trailing whitespace
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

After 2nd VSG fix (no errors)

```
entity adder_tree is
  generic (
    IN_WIDTH  : integer := 14;
    OUT_WIDTH : integer := 15
  );
  port (
    Q : out   std_logic_vector((OUT_WIDTH - 1) downto 0)
  );
end entity adder_tree;

```

jeremiah-c-leary, 2022-01-31:
Hey @imd1,

I found the issue and pushed a fix to the issue-682 branch.  If you get a chance, could you check it out on your end?  Othewise I will plan to merge this into master after 02/07/2022.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-02-05:
Hey @imd1,

I just merged this to master for inclusion in the next release.

--Jeremy

closed 2022-02-05 by jeremiah-c-leary