jeremiah-c-leary/vhdl-style-guide #683: VSG gives error on PORT definition case when using a 'subtype or record.

rugebiker, 2021-11-11:
**Environment**
Master branch

**Describe the bug**
Hi, Consider the case that I have the following rules:

- Port definitions should be uppercase
- Port definitions should keep case consistency
- Signal definitions should be lowercase
- Signal definitions should keep case consistency

Now lets imagine I have two port definitions as follows:

```
PORT1 : in std_logic_vector(7 downto 0);
PORT2 : in record1;
```

now lets imagine that I have two signals declared as follows:

```
signal signal1 : PORT1'subtype;
signal signal2 : PORT2.record_field;
signal signal3 : std_logic_vector(PORT1'range);
```

If I run VSG over this file, then it gives me an error that in the declaration of signal1, PORT1 should be lowercase, and in the declaration of signal2 it tells me that PORT2 should be also lower case. Both of these statements are wrong. If as an experiment I put them in lowercase and run VSG again, then it gives me an error saying that they should be uppercase, so this keeps in kind of an infinite loop.

signal3 never gives an error by VSG, which is working correctly. I just added this declaration on the example to show that here VSG works correctly.

**Expected behavior**
VSG shouldn't give erros on the example above.

Thanks!
--Ruben

jeremiah-c-leary, 2022-02-04:
Hey @rugebiker,

So the rule `signal_011` is an old rule before the consistent case rule was created.  Disabling the rule is one option, but I am thinking it should be removed.

Do you have a preference?

--Jeremy

jeremiah-c-leary, 2022-02-06:
Hey @rugebiker,

Actually, it looks like I will need to update my parser to handle this correctly.

--Jeremy

rugebiker, 2022-02-11:
Hi @jeremiah-c-leary  :)

The branch you mentioned doesn't exist anymore, but saw that it was merged to master, so I pulled master and remove/replaced the deprecated rules, and now this issue seems solved!

Thanks a lot :)
--Ruben

jeremiah-c-leary, 2022-02-11:
Awesome!!! I will close this issue.

--Jeremy

closed 2022-02-11 by jeremiah-c-leary