jeremiah-c-leary/vhdl-style-guide #684: VSG reports success while silently leaving violations unfix (1)

imd1, 2021-11-12:
**Environment**
VSG 3.3.2, Windows 10

**Describe the bug**
Original code:
```

entity an_entity is
  generic (
    runner_cfg : string;
    tb_path    : string   
  );
end  ;

architecture tb of an_entity is

begin

end architecture;
```

After running VSG in fix mode, it reports successful fixing and the file becomes:

```

entity an_entity is
  generic (
    RUNNER_CFG : string;
    TB_PATH    : string
  );
end entity   an_entity;

architecture tb of an_entity is

begin

end architecture tb;

```

Running VSG a second time in check only mode reports remaining violations to fix (which it can if I run in fix mode):
```
================================================================================
File:  /path/to/file.vhd
================================================================================
Phase 2 of 7... Reporting
Total Rules Checked: 206
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  entity_013                | Error      |          7 | Reduce spaces between *entity* keyword and the entity simple name to a single space.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

```

Having to run a tool twice when it reports success on the first run does not give a good impression.

**Additional context**
Add any other context about the problem here.

jeremiah-c-leary, 2021-12-04:
I suspect the space between the `end` and semicolon is the culprit.

jeremiah-c-leary, 2021-12-04:
Hey @imd1,

I have a preliminary fix for this on the issue-684 branch.  If you get a chance could you check it out.

I still need to do some refactoring before I am ready to merge this in.

Thanks,

--Jeremy

jeremiah-c-leary, 2021-12-05:
Hey @imd1,

I finished my refactoring of this issue.  When you get a chance, could you check it out, or let me know if I should just merge it in?

Thanks,

--Jeremy

jeremiah-c-leary, 2021-12-12:
Hey @imd1,

I am going to merge this to master for the 3.4.0 release.  If this is still a problem we can reopen this issue.

Thanks,

--Jeremy

closed 2021-12-12 by jeremiah-c-leary