jeremiah-c-leary/vhdl-style-guide #685: VSG reports success while silently leaving violations unfixed (2)

imd1, 2021-11-12:
**Environment**
VSG 3.3.2, Windows 10

**Describe the bug**
Original code:
```

library work;
use work.tb_utilities.all;
use work.tb_wait_clock_package.all;
use work.random_pkg.all;

use work.dsd_types.all;
--use work.channel_types.all;

```

After running VSG in fix mode, it reports successful fixing and the file becomes:

```

library work;
  use work.tb_utilities.all;
  use work.tb_wait_clock_package.all;
  use work.random_pkg.all;
  use work.dsd_types.all;
--use work.channel_types.all;
```

Running VSG a second time in check only mode reports remaining violations to fix (which it can if I run in fix mode):
```
================================================================================
File: /path/to/file.vhd
================================================================================
Phase 4 of 7... Reporting
Total Rules Checked: 367
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  comment_010               | Error      |          7 | Indent level 1
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

Having to run a tool twice when it reports success on the first run does not give a good impression.

**Additional context**
Add any other context about the problem here.

jeremiah-c-leary, 2021-12-05:
I suspect it has something to do with the indent logic, or more specifically the setting of the indent after the code has been modified.

jeremiah-c-leary, 2021-12-05:
Yeah,
It is a confluence of setting the indent and removing the blank line.  Library statements are hard to indent.  When I remove the blank line, I do not redo the indents.  So after the fix is performed, the indent is not updated then when the final analyze is done it passes.  When you re-run it, it fails.

I would rather not redo the indent after each phase, so I will need to think of the best way forward.

jeremiah-c-leary, 2021-12-05:
Hey @imd1,

I pushed a fix for this to the issue-685 branch.  I still need to do some refactoring, but it is appears to be functional.  If you get a chance could you double check, or let me know if I can just merge it.

--Jeremy

jeremiah-c-leary, 2021-12-18:
Hey @imd1,

I am going to merge this into trunk so I can release version 3.4.0.  If there are still issues we can address them as they come up.

--Jeremy

closed 2021-12-18 by jeremiah-c-leary