jeremiah-c-leary/vhdl-style-guide #686: Generic prefix suffix rules not applied to generic map clauses

imd1, 2021-11-15:
**Environment**
VSG 3.3.2, Centos 7.6.1810

**Describe the bug**

The generic prefix suffix rules (020, 600) do not seem to apply to generic_map clauses

imd1, 2021-11-15:
After further thought, it makes sense adding a new pair of rules for this

imd1, 2021-11-15:
Implemented this on my fork
* https://github.com/imd1/vhdl-style-guide/pull/1

Not sure how you want to proceed

jeremiah-c-leary, 2021-12-05:
Do we worry about generics that do not have suffixes/prefixes.  For example, let's say we use a piece of IP that we can not modify then if we change the prefix/suffix on the instantiation then it will not compile.

imd1, 2021-12-12:
> Do we worry about generics that do not have suffixes/prefixes

Well, you have generic_020 etc for this, so ?

imd1, 2021-12-12:
> For example, let's say we use a piece of IP that we can not modify then if we change the prefix/suffix on the instantiation then it will not compile.

True, but that's a corner case and you would exclude the rule using code tags. In general most instantiations are your own code and if you've set generic naming rules up, you'd want to be able to apply them to generic maps?

jeremiah-c-leary, 2022-02-12:
Could you make a pull request for this and I can look at it and merge?

Thanks,

--Jeremy

imd1, 2022-02-14:
> Could you make a pull request for this and I can look at it and merge?
> 
> Thanks,
> 
> --Jeremy

Done, although it looks a bit "stale"

imd1, 2022-02-14:
https://github.com/jeremiah-c-leary/vhdl-style-guide/pull/743

jeremiah-c-leary, 2022-04-23:
Hey @imd1,

I had to merge your pull request locally and address some merge conflicts.  I pushed the updates to the imd1-issue-686 branch.  If you have a chance could you check it out.

I will plan to merge this to master by Apr 29th if I do not hear from you by then.

Thanks,

--Jeremy

PR #756 (2022-04-30): Imd1 issue 686
Resolves #686
closed 2022-04-30 by jeremiah-c-leary