jeremiah-c-leary/vhdl-style-guide #687: indentWidth of 0 stays an error

timkruse, 2021-11-18:
**Environment**
VHDL Style Guide (VSG) version: 3.3.2
Git commit SHA: 8ce9b1d

Installed on Ubuntu 20.4 via `pip install vsg`

**Description**
The rule `indentWidth` stays an error if set to `0` independent of being defined as a global or a specific rule. Fixing `vsg -c .vsg-format 
 -f test.vhd --fix` does work correctly, though.

**To Reproduce**
Given the following code (which is indented correctly)

```vhdl
library ieee;
use ieee.std_logic_1164.all;
```
and this configuration

```json
{
    "rule": {
        "global": {
            "indentSize": 0
        }
    }
}
```

and run with

```bash
vsg -c .vsg-format -f test.vhd
```

Gives the following output

```
================================================================================
File:  test.vhd
================================================================================
Phase 4 of 7... Reporting
Total Rules Checked: 370
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  library_008               | Error      |          2 | Indent level 1
----------------------------+------------+------------+--------------------------------------
```

Fixing the issue via the `--fix` flag does correctly indent the code but the error persists during subsequent checking calls `vsg -c .vsg-format -f test.vhd`

jeremiah-c-leary, 2021-12-05:
Hey @timkruse,

I pushed an update for this issue to the issue-687 branch.  If you get a chance, could you try it out and let me know if it fixes the issue on your side?

Thanks,

--Jeremy

timkruse, 2021-12-06:
I just checked out the issue-687 branch but it did not fix the issue, unfortunately. Can I support you with any further information from me?

jeremiah-c-leary, 2021-12-06:
I noticed you had a global indent size of 0, are you seeing failures on lines other than the `use_clause`?

timkruse, 2021-12-06:
No, it is only the case for the `use_clause`. I should have mentioned that the issue originated from having this rule applied

```json
"library_008": {
    "indentSize": 0
},
```

jeremiah-c-leary, 2021-12-06:
hmm...that is strange.
I copied your example code:
```vhd
  1 library ieee;
  2 use ieee.std_logic_1164.all;
  3
```
...and used this configuration:
```json
  1 {
  2     "rule": {
  3         "global": {
  4             "indentSize": 0
  5         }
  6     }
  7 }
```
I run this command to test it:
```mono
cp original.vhd fixed.vhd;vsg -f fixed.vhd --fix -c config.json
```
and I get:
```mono
================================================================================
File:  fixed.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 570
Total Violations:    0
  Error   :     0
  Warning :     0
```
I think my inputs match yours, but we get different results.

I attached a zip file with config and test file, could you run it on your end?
[issue_687.zip](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/7663306/issue_687.zip)

timkruse, 2021-12-07:
The issue only appears without the `--fix` flag. With that flag it indents correctly and does not raise errors.

jeremiah-c-leary, 2021-12-07:
I just tried this command on `fixed.vhd`:
```mono
vsg -f fixed.vhd -c config.json
```
and it did not report an error:
```mono
================================================================================
File:  fixed.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 570
Total Violations:    0
  Error   :     0
  Warning :     0
```
Any chance you are not running on the branch?

If you do a git log you should see these two commits on the branch:
```mono
commit c255f3c0d8aa6be0aa6e896f95229ef648fad91d (HEAD -> issue-687, origin/issue-687)
Author: Jeremiah Leary <jeremiah.c.leary@gmail.com>
Date:   Sun Dec 5 15:39:36 2021 -0600

    Updates for issue #687

    Refactoring to improve readability.

commit 280ca3b1918dc7667ef5bae0a2c283a4f4631b2e
Author: Jeremiah Leary <jeremiah.c.leary@gmail.com>
Date:   Sun Dec 5 15:30:24 2021 -0600

    Updates for issue #687

    Fixing issue with use_clause when indentSize is set to 0.

      1)  Added test to expose the issue
      2)  Updated token_indent base rule to handle indentSize of 0
```

timkruse, 2021-12-07:
Well, I am on the right branch but I missed to install the repository. I was using the `[repo]/bin/vsg` file. 

After installing the `issue-687` branch

```bash
$ python setup.py install
$ /usr/local/bin/vsg --version
VHDL Style Guide (VSG) version: 3.3.0.dev50
Git commit SHA: gc255f3c0
```

it works correctly.

Thank you very much!

closed 2021-12-07 by timkruse