jeremiah-c-leary/vhdl-style-guide #691: when-else statement causes a crash when used in a function for a variable

azdem, 2021-11-29:
$ vsg --version
VHDL Style Guide (VSG) version: 3.3.2
Git commit SHA: 8ce9b1d

Code snippet:
```
function init_mem return ramType3 is 
    variable hpp : std_logic;
...
begin
    hpp := '1'  when (pf_vlan2x_tci(3 downto 0) >= x"A" and pf_vlan2x_tci(3 downto 0) <= x"F") else '0';
```
Last line causes VSG to crash with message: Expecting when instead of else.

Regards

jeremiah-c-leary, 2021-12-01:
Hey @azdem,

It turns out my parser was classifying the variable_assignment as a signal_assignment.  It detected the less than equal (<=) and assumed it was a signal assignment operator.  I put in a fix as part of my investigation, but I need to enhance my tests to check for this case.

I will see if I can get a fix out this weekend.

--Jeremy

jeremiah-c-leary, 2021-12-04:
Hey @azdem,

I just pushed a fix for this to master.  I believe I need to do a hot fix for several other items, but you can pull down master if want to double check I fixed it.

--Jeremy

jeremiah-c-leary, 2021-12-11:
Hey @azdem,

I am planning to release 3.4.0 this weekend and I would like to get this issue in the release.  So I will close this issue.  If you still experience any problems we can re-open this ticket or create another one.

Thanks,

--Jeremy

closed 2021-12-11 by jeremiah-c-leary