jeremiah-c-leary/vhdl-style-guide #699: sequential_004 should allow alignment configuration

staerz, 2021-12-16:
**Is your feature request related to a problem? Please describe.**
This rule checks the alignment of multiline sequential statements but doesn't allow any configuration.

Other rules like concurrent_009 do allow such a configuration (what is the exact difference between sequential_004 and concurrent_009?).

Similar to #499.

**Describe the solution you'd like**
I'd like to be able to configure the alignment to follow a default 2-spaces indentation.

**Describe alternatives you've considered**
Not caring.

jeremiah-c-leary, 2022-01-01:
Would having the same configuration options as `concurrent_009` be sufficient?

staerz, 2022-01-03:
Hi @jeremiah-c-leary,

I would say that this would make sense.

Again, as mentioned in the issue description, I'm actually not super sure to understand the difference between `sequential_004` and `concurrent_009`. Is it that a simple signal assignment statement would classify as `sequential_004` and as soon as I add a `when` condition, it turns into `concurrent_009`? Or does the first require to be within a `process` statement?

The key of my question is if they couldn't be the same rule.

Please note that I also updated the issue description: While checking I found a minor bug in the documentation.

jeremiah-c-leary, 2022-02-13:
Hey @staerz,

> Again, as mentioned in the issue description, I'm actually not super sure to understand the difference between sequential_004 and concurrent_009. Is it that a simple signal assignment statement [...]

After researching a little, it looks like my multiline rules are a bit a of a mess.

[image]

Where:

* multiline_alignment_between_tokens = base rule which multiple rules can extend
* unique = implementation is only in that rule
* alignment = rule performs alignment
* structure = rule modifies the format of the code

I might be able to merge the options `wrap_at_when`, `align_when_keywords` and `align_else_keywords` into the `multiline_alignment_between_tokens` base rule.  Then I could continue to extend the base rule to cover additional LRM Entities.

I need to ponder this for a bit.

--Jeremy

staerz, 2022-02-14:
Yes, I would agree to unification here: All these are structural language elements where the same rules can apply (and should).

If you can hence manage to define a base rule and derive just the particular rules basically by the LRM keyword kicking in to catch up that base rule, that would be most consistent.

jeremiah-c-leary, 2022-10-08:
Hey @staerz ,

Since the updates for this issue are the same for variable_assignments, I was going to do updates for this issue on issue #549.

We can close this issue when we close the other issue.

--Jeremy

staerz, 2022-10-11:
Yes, that sounds very fair to me and also funnels the discussion to one issue only.

PR #839 (2022-10-20): Issue 549a
Resolves #549 
Resolves #699
closed 2022-10-20 by jeremiah-c-leary