jeremiah-c-leary/vhdl-style-guide #705: Add configuration for constant_014

staerz, 2022-01-20:
**Is your feature request related to a problem? Please describe.**
"I'm always frustrated when" things look ugly or inconsistent.

**Describe the solution you'd like**
The current (non-configurable) configuration for [constant-014](https://vhdl-style-guide.readthedocs.io/en/latest/constant_rules.html#constant-014) is the following:

Violation:
```vhdl
constant width : integer := a + b +
  c + d;
```
Fix:
```vhdl
constant width : integer := a + b +
                            c + d;
```

I find the indentation very annoying as its position depends the name of the constant and its type (definition). Instead I would prefer a (to my opinion) much more consistent indentation like this:
Fix:
```vhdl
constant width : integer :=
  a + b +
  c + d;
```
The difference is:
- the value given, **if spread over multiple lines** is started given on the **next line** only, with the default indentation (I use 2 spaces here, but that should be taken from the global indentation setting) and
- any following line should be indented in the exact same manner.

An additional check **could** be that the first valid character of a new line must be a non-arithmetical operator, such that the following would also be a violation:
```vhdl
constant width : integer :=
  a + b
  + c + d;
```

I hence suggest to add a configuration for the indent of this rule.

**Describe alternatives you've considered**
The alternative is to disable the rule.

**Additional context**
This occurred to me when trying the following code:
```vhdl
      --! Index (lower bound) to where the valid is mapped on the output monitor
      --! It is the accumulative sum of lolli and patgen links up to this tile
      constant LINK_IDX_N : natural :=
        ite(tile=2, G_FEB2_LINK_LOLLI_TILE1_N + G_FEB2_LINK_PATGEN_TILE1_N, 0) +
        ite(tile=3, G_FEB2_LINK_LOLLI_TILE2_N + G_FEB2_LINK_PATGEN_TILE2_N, 0) +
        ite(tile=4, G_FEB2_LINK_LOLLI_TILE3_N + G_FEB2_LINK_PATGEN_TILE3_N, 0);
```
Here, the aim is to render the lengthy definition readable and hence split the sum into a line for each contributor. Here, `ite` is an "if-then-else"-function and the code resides in a `tile`-defining `generate` statement.

Note that this issue may in strucutre also apply to default values for signals (although one shouldn't use default values for signals in general).

jeremiah-c-leary, 2022-01-26:
It sounds like this would be similar to the other alignment rules with the `align_left` option.

staerz, 2022-01-26:
Yes, absolutely. So it is actually all about consistency!

jeremiah-c-leary, 2022-02-01:
Hey @staerz,

I pushed an update for this to the `issue-705` branch.  When you get a chance, could you check it out on your end.  I will plan to push this to master after 02/08/2022 if I do not hear back from you.

Thanks,

--Jeremy

staerz, 2022-02-01:
Hi @jeremiah-c-leary,

it seems to work.

Note that I saw your updates of documentation only from the raw changes and spotted a thing.

First, when simply duplicating the configuration from `constant_012` to `constant_014`, i.e.
```json
    "constant_014":{
      "align_left":true,
      "align_paren":false
    },
```
I get the following VSG failure:

```
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/apply_rules.py", line 98, in apply_rules
    oRules.check_rules(
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rule_list.py", line 221, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rules/multiline_alignment_between_tokens.py", line 127, in analyze
    if indents_match(dActualIndent[iLine], dExpectedIndent[iLine]):
UnboundLocalError: local variable 'dExpectedIndent' referenced before assignment
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==3.0.0.dev20', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/__main__.py", line 138, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 868, in next
    raise value
UnboundLocalError: local variable 'dExpectedIndent' referenced before assignment
```

I saw that you opted for making those configurations a "boolean", but taking values "yes" and "no" (similar to what's done with config for `concurrent_009` for example.
Yes, no problem making it "yes" and "no", but then let's please be consistent: the documentation should in no case refer to them as "boolean", but rather "string".

Note that I browsed the documentation a bit and found it inconsistent when comparing to the following two which have similar attributes:
- https://vhdl-style-guide.readthedocs.io/en/latest/configuring_conditional_expressions.html
-  https://vhdl-style-guide.readthedocs.io/en/latest/configuring_multiline_indent_rules.html

I think here some cleanup would be useful, making sure that the same attribute names are of the same type and the configuration values then also match up. Now it's a mixture of boolean, string and True, False, 'Yes', "No".

I guess I should now also update my configuration of `constant_012` from
```json
    "constant_014":{
      "align_left":true,
      "align_paren":false
    },
```
to
```json
    "constant_014":{
      "align_left":"yes",
      "align_paren":"no"
    },
```
!? ... or not, if you decide to keep them all boolean :wink:

jeremiah-c-leary, 2022-02-03:
> Now it's a mixture of boolean, string and True, False, 'Yes', "No".

That must have happened when I was refactoring constant_012.

I will align `constant_012` with the others, and fix the documentation.

jeremiah-c-leary, 2022-02-03:
Hey @staerz,

I just pushed an update where `constant_012` takes `yes` and `no` strings.  I also updated the Configuring Multiline Indent Rules documentation.

Thanks for catching the inconsistency.

--Jeremy

staerz, 2022-02-03:
Hi @jeremiah-c-leary,

alright, great! We both seem to like consistency :+1: 

I think I'll only be able to look at the documentation once it's on master (or is there any way I can browse the readthedocs on a particular branch ... is that even created?), so let's assume it's good.

In case I would find anything later, you know what would happen and it's not a problem :smile: 

So please merge and we can close this issue!

jeremiah-c-leary, 2022-02-04:
Hey @staerz,

> We both seem to like consistency

Yes...I like consistency, I just struggle to be consistent :grin:.

> I think I'll only be able to look at the documentation once it's on master

The documentation is under the `docs` directory.

>  So please merge and we can close this issue!

Sweet!!

closed 2022-02-04 by jeremiah-c-leary