jeremiah-c-leary/vhdl-style-guide #709: Provide inverses of generic_010 and port_014

ktbarrett, 2022-01-27:
**Is your feature request related to a problem? Please describe.**
I am working on trying to replicate our companies internal VHDL linter with VSG and our style requires that generic and port statements in entity declarations always have the trailing parens on the same line as the last element. generic_010 and port_014 enforce the opposite rule.

**Describe alternatives you've considered**
I can turn these checks off, but it doesn't enforce the opposite rule.

**Describe the solution you'd like**
Add a new generic and port rule to enforce trailing parens in generic and port statements in entity declarations to be on the same line as the last element (with no spaces after the previous token).

jeremiah-c-leary, 2022-01-30:
Hey @ktbarrett,

I should be able to add an option to the existing rules to enforce the opposite.

jeremiah-c-leary, 2022-01-30:
Hey @ktbarrett,

I pushed an update for this issue to the issue-709 branch.  When you get a chance could you check it out and let me know what you think?  I will plan to push this to master after 2/6/2022 if I do not hear from you.

Regards,

--Jeremy

ktbarrett, 2022-02-01:
@jeremiah-c-leary The current process at my place of work doesn't allow us to test code. I will have to request approval of the new version once it's released. I'm looking at #713 and I see no mention of generic_010 and port_014 as mentioned in the OP. Is that covered?

jeremiah-c-leary, 2022-02-01:
Hey @ktbarrett,

>  I'm looking at #713 and I see no mention of generic_010 and port_014 as mentioned in the OP. Is that covered?

Looks like I missed those.  I will add them.

> The current process at my place of work doesn't allow us to test code. I will have to request approval of the new version once it's released.

After adding generic_010 and port_014, I can make a release 3.7.0.  If there are additional changes then I can update and re-release.

How long would it take to get a new version in?

jeremiah-c-leary, 2022-02-02:
Hey @ktbarrett,

I updated `generic_010` and `port_014`.  Let me know how you want to proceed.

--Jeremy

jeremiah-c-leary, 2022-02-05:
Hey @ktbarrett,

I am going to push this to master for the next release so you can test it out.

--Jeremy

closed 2022-02-05 by jeremiah-c-leary