jeremiah-c-leary/vhdl-style-guide #710: Support custom alignment of in/out/inouts

ktbarrett, 2022-01-27:
**Is your feature request related to a problem? Please describe.**
Our internal style has the in and out part of port direction aligned, see an example below.

```vhdl
port (
    abc   : in    std_logic;
    def   :   out std_logic;
    ghi   : inout std_logic);
```

**Describe alternatives you've considered**
I can disable port_005 and port_008, but this doesn't enforce the aforementioned style.

**Describe the solution you'd like**
The rules list "must be 1 space before port name", it might be easy to be able to modify these rules to allow the user to specify the number of spaces. And it would be necessary to split the rule port_005 between in, out, and inout.

jeremiah-c-leary, 2022-01-30:
Hey @ktbarrett,

I pushed an update for this issue to the issue-710 branch.  When you get a chance, could you check it out and let me know what you think?  I will plan to merge this into master after 2/6/2022 if I do not hear from you by that time.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-02-05:
Hey @ktbarrett,

I am going to push this to master for the next release so you can test it.

--Jeremy

closed 2022-02-05 by jeremiah-c-leary