jeremiah-c-leary/vhdl-style-guide #719: Misleading Solution for comment_100

staerz, 2022-02-01:
**Environment**
VSG 3.6.0-4-g45ef2a89 (on branch for testing issue-705)

**Describe the bug**
The given Solution for `comment_100` is misleading, likely due to a typo or similar:

The code
```vhdl
    --! Configuration Clock @todo find correct pin assignment
    --s10_osc_clk_1       : in    std_logic;
```
Produces the following VSG output:
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  comment_100               | Error      |         48 | Change "--s" to "-- 1"
----------------------------+------------+------------+--------------------------------------
```

The solution indicated would translate to "Replace the s by a space", but the intention is to "Insert a space after '--'".

**To Reproduce**
Steps to reproduce the behavior:
1. Have any code line start as a comment ('--') and directly after start some text
2. Run VSG on it
3. See error

**Expected behavior**
Fixing the provided solution. I would recommend changing the text in the solution to "Insert a space after '--'".

**Additional context**
It's a minor thing, but clearly confusing :wink:

jeremiah-c-leary, 2022-02-04:
That is confusing. 

I will put a check for the solution into my test.

jeremiah-c-leary, 2022-02-04:
Hey @staerz,

I pushed a fix for this to the issue-719 branch.  It turns out I was using the wrong index in my solution function. This resulted in picking the character after the one I wanted.

I want you can check it out or I can just push it to master.

--Jeremy

staerz, 2022-02-04:
Hi @jeremiah-c-leary,

yes, I confirm the fix works. The complaint now is
```
  comment_100               | Error      |         48 | Change "--s" to "-- s"
```
which is catching the right characters and also the `--fix` option does (still) the correct thing.

Approved!

Please go ahead and merge and close the issue!

jeremiah-c-leary, 2022-02-05:
Sweet!

closed 2022-02-05 by jeremiah-c-leary