jeremiah-c-leary/vhdl-style-guide #720: Documentation: Broken link(s)

staerz, 2022-02-01:
**Environment**
Referring to latest master as observed on readthedocs.

**Describe the bug**
Some links to the documentation are broken, or are not pointing to the intended subsection.

E.g. pointing to the configuration of blank lines (e.g. in `block_20x` or `subprogram_body_20x`) the used link is `https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#configuring-blank-lines` while it should be `https://vhdl-style-guide.readthedocs.io/en/latest/configuring_blank_lines.html`, unless it is intended to link to the overview page of configuration. 

It is likely that this observed link mismatch also is true for other links of that kind pointing to the configuration section.

**To Reproduce**
See the documentation.

**Expected behavior**
It would, for a better user experience, be better to directly link to the particular configuration of the rule in question than to point to the configuration overview.

**Additional context**
It is likely a non-intended feature by how links are possibly auto-generated.
Ideally all links should be checked in some automated way when attempting to address this issue.

jeremiah-c-leary, 2022-02-05:
> Some links to the documentation are broken, or are not pointing to the intended subsection.

I believe this occurred after I changed the configuration documentation to be a toc instead of included.

I updated all the links by changing them to a "substitution" in Restructured Text.
```mono
  1
  2 .. |configuring_blank_lines_link| replace::
  3    Refer to :ref:`configuring-blank-lines` for more information.
```
...and then called out the label in the documentation:
```mono
157 library_007
158 ###########
159
160 |phase_3| |error| |blank_line|
161
162 This rule checks for blank lines or comments above the **use** declaration.
163
164 |configuring_blank_lines_link|
165
```
Now I can have all my links be consistent.

>For a user it would be easier if it were in alphabetic order (the part "Configuring ...").

I agree and I arranged them in alphabetical order.

I pushed an update for this to master so you could see the results on read the docs.

Let me know what you think.

--Jeremy

staerz, 2022-02-07:
Hi @jeremiah-c-leary,

that looks pretty slick now! I did test a few random rules and clicked on the linked configuration and always ended up where I was supposed to. Great!

Also the new order of the main config page looks so much more organised now!

Thanks! Approved. Feel free to close the issue :smile:

jeremiah-c-leary, 2022-02-07:
Sweet!!

closed 2022-02-07 by jeremiah-c-leary