jeremiah-c-leary/vhdl-style-guide #721: False negative for block_comment_001

staerz, 2022-02-01:
**Environment**
3.6.0-4-g45ef2a89

**Describe the bug**
A false negative, i.e. an error report while everything is correct in what seems a particular configuration.

The following code
```vhdl
<pseudo:
some commented lines

a block comment

another block comment

library call

entity definition with ports only (no generics)
end pseudo>

--! Implementation of the avst_packetiser
architecture behavioral of avst_packetiser is

  --! Intermediate signal for incoming data
  signal data_r  : std_logic_vector(avst_stream_i.data'range);
  --! Intermediate signal for incoming valid
  signal valid_r : std_logic;
  --! Intermediate signal for incoming start-of-packet
  signal sop_r   : std_logic;

begin

------------------------------<-    80 chars    ->------------------------------
--! FPGA RX to Avalon-ST TX interface
--! to make it really stable, two cycles are required to produce the eof signals
--------------------------------------------------------------------------------

```
produces this:
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  block_comment_001         | Error      |         61 | Change block comment header to : -----------------------------<-    80 chars    ->-----------------------------
```

The proposed solution (and what is also applied with the `--fix` option) is to remove 1 '-' each, left and right, which then actually malformats the block comment.

Seems to be a calculation bug being confused by some indent counter or something.

Note the configuration for block comments:
```json
    "block_comment_001":{
      "disable": false,
      "header_left_repeat": "-",
      "header_string": "<-    80 chars    ->",
      "header_right_repeat": "-",
      "max_header_column": 80,
      "allow_indenting": false,
    },
    "block_comment_002":{
      "disable": false,
    },
    "block_comment_003":{
      "disable": false,
      "max_footer_column": 80,
      "allow_indenting": false,
    },
```

**To Reproduce**
Steps to reproduce the behavior:
1. Have code as above
2. Run VSG with the configuration given
3. See error

**Expected behavior**
I do not expect VSG to complain here.

**Additional context**
Coming still from VSG 3.0.0, not having followed minor version updates so far (I have some catching up to do ... expect more reports).

jeremiah-c-leary, 2022-02-03:
Hey @staerz,

It looks like the rule was taking the current indent into account when calculating what the header should look like.

I pushed a fix for this to the issue-721 branch.  When you get a chance could you check it out on your end.

Thanks,

--Jeremy

staerz, 2022-02-03:
Hi @jeremiah-c-leary,

I confirm that the provide patch fixes the problem and looking a bit at the code, the solution looks good to me (and was like I expected :wink: ).

Please go ahead and merge and then the pleasure of closing this issue is all yours!

jeremiah-c-leary, 2022-02-04:
> Please go ahead and merge and then the pleasure of closing this issue is all yours!

Sweet.  It will be in the next release.

closed 2022-02-04 by jeremiah-c-leary