jeremiah-c-leary/vhdl-style-guide #722: Incomplete case rules w.r.t. blank lines

staerz, 2022-02-01:
**Is your feature request related to a problem? Please describe.**
Being surprised by the new rule `case_200`, I did some testing and found an inconsistency (the hole in the matrix) regarding the `case` statement and the rules VSG currently implements on them, esp. `case_008` and `case_009`.

The "problem" is hence incompleteness.

**Describe the solution you'd like**

The following code
```vhdl
        case pre_locked_state is

          when IDLE =>

            if bcr_s = '1' then
              bcr_expect_rst   <= '1';
              pre_locked_state <= FIRST;
            else
              pre_locked_state <= IDLE;
            end if;

          when FIRST =>

            if bcr_expect = '1' then
              pre_locked_state <= CHECKED;
            else
              pre_locked_state <= FIRST;
            end if;
          when CHECKED =>

            if (bcid_first_r(1) xor bcr_expect) then
              pre_locked_state <= IDLE;
            else
              pre_locked_state <= CHECKED;
            end if;

        end case;
```
passes VSG without problems (using default configuration).

Note that although `case_008` and `case_009` check for a blank line after `case ... is` and before `end case`, there is no rule that checks for a blank line before entering the new case (indicated by the `when` keyword).

It is possibly a matter of taste and habit, but I would argue that if one leaves a blank line after the `case ... is`, that this would imply to leave a blank line **above** the `when ... =>` section of **all cases**.
Note that for the first `when` section, "below the `is` keyword" is equivalent to "above the `when ... =>` section.

Hence, rather than introducing a new rule, I would promote to amend the current rule `case_008`. Instead of checking "below the `is` keyword", rather "above the `when ... =>` sections" should be tested. This would include the current definition and capture all cases.

**Describe alternatives you've considered**
An alternative is to not care about this formatting.

An alternative to amending `case_008` would be to create a new rule similar to `case_200` **but** this has the potential to allow mutually exclusive configurations in a setup (which I'd always intend to avoid by design).

**Additional context**
Happily checking out the latest VSG version...

jeremiah-c-leary, 2022-02-01:
Hey @staerz,

Looks like I have every other blank line rule except for that one.

> An alternative to amending case_008 would be to create a new rule similar to case_200 but this has the potential to allow mutually exclusive configurations in a setup (which I'd always intend to avoid by design).

I see you point on this.  The only thing I worry about is coments.

```vhdl
  case state is
    -- Comment
    -- Comment
    when state1 =>
```

I would use the `previous_line` rule we created to handle blank lines above the `when` keyword.  This would allow for comments, etc...

It looks like the `allow_comment` and `require_comment` would handle those cases though.  However that does not address blank lines after the `is` keyword.

I do have a rule that removes "excessive" blank lines.  I could have it enforce no more than a single blank line after it:

```vhdl
case state is

  --Comment
  when state1 =>
```
would be fixed to:
```vhdl
case state is

  --Comment
  when state1 =>
```

I think that would remove the conflicts.

So I would propose the following:

  1)  Remove rule `case_008`
  2)  Add rule `case_201` to check for "excessive" blank lines below the `is` keyword
  3)  Add rule `case_202` to check for blank lines above the `when` keyword using the `previous_line` rule.

staerz, 2022-02-02:
Hi @jeremiah-c-leary,

I see, I didn't think about comments. We seem to be quite complementary when it comes to defining rules :+1: 

Yes, I agree, every `when` could come with a comment and the very same rules should apply here as for all the other "may have a comment here"-rules.

When it comes to your rule to remove excessive blank lines, I'm wondering if that should be tied to any particular syntax keyword. Wouldn't it make sense to apply this as a global rule which would check **anywhere** in the code for 2 (or more) blank lines? I may be biased as I cannot imagine a style which would allow double blank lines anywhere.

But OK, in the end, what you propose does the trick (I'm really just wondering if we need another rule, but you know VSG best).

jeremiah-c-leary, 2022-02-04:
Hey @staerz,

> When it comes to your rule to remove excessive blank lines, I'm wondering if that should be tied to any particular syntax keyword. Wouldn't it make sense to apply this as a global rule which would [...]

That is a good point.  Let me think about it.

--Jeremy

jeremiah-c-leary, 2022-02-04:
Hey @staerz,

I created a new issue #727 to remove excessive blank lines over the entire file.

Meanwhile, I pushed an update for this issue to branch issue-722.  Let me know what you think.

--Jeremy

staerz, 2022-02-04:
Hi @jeremiah-c-leary,

I confirm the proper behaviour now.

Using `git diff` to look at changes, I found "after the **context** keword" for the documentation of `case_202`. 2 things here: 1) typo "keyword", 2) typo "case".

I guess that it is intentional to introduce `case_202` here while #727 is not yet done, but if you ask me, save it if you anyway plan to work on it soon.

jeremiah-c-leary, 2022-02-05:
> I guess that it is intentional to introduce case_202 here while https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/727 is not yet done, but if you ask me, save it if you anyway plan to work on it soon.

That was my intent, but thinking about it now I would just be removing `case_002` shortly anyway.  I would also have to maintain a deprecated rule for it.  That is just extra work.  It is easy enough to remove the rule.

jeremiah-c-leary, 2022-02-07:
Sweet!!

closed 2022-02-07 by jeremiah-c-leary