jeremiah-c-leary/vhdl-style-guide #723: Comments after "end if;"

sibeov, 2022-02-02:
**What is your question?**
I use comments after `end if;` to have make it more logical as to what conditional one is checking. I was having trouble finding the correct rule which disables the comment to be indented "all the way to the right".

I disabled the rule for inline comments and hoped that it would do it (comment_011).

Any suggestions? 

Example:
```vhdl
-- What I try to do
if something_is_valid then

end if; -- Something valid

-- What happens when I indent
if something_is_valid then

end if;                                                             -- Something valid

```

Kind Regards
sibeov

P.S.
Fantastic project. Many thanks!

jeremiah-c-leary, 2022-02-02:
Hey @sibeov,

> Fantastic project. Many thanks!

Thanks for the kind words.

> I disabled the rule for inline comments and hoped that it would do it (comment_011).

comment_011 is already disabled, which you can check with the `-rc` option:
```mono
$ vsg -rc comment_011
```
```json
{
  "rule": {
    "comment_011": {
      "indentSize": 2,
      "phase": 1,
      "disable": true,
      "fixable": true,
      "severity": "Error"
    }
  }
}
```

> I was having trouble finding the correct rule which disables the comment to be indented "all the way to the right".

We need to keep in mind two concepts to find the rule which is moving the comment to the right:

  1)  indenting versus alignment
  2)  How alignment rules are classified

Indenting Versus Alignment
-------------------------------

In VSG uses the term indenting to refer to whitespace at the beginning of the line and "aligning" to all other whitespace.  So in your case we need to look for an alignment rule.

How Alignment Rules are Classified
----------------------------------------

Alignment across multiple VHDL statements are classified with rules that are within for the element that contains the multiple VHDL statements.  For example:

```vhdl
architecture rtl of fifo is

  signal sig1 : integer;        -- Comment 1
  constant con1 : natural := 0;   -- Comment 2

begin

  process

    signal sig2 : integer;   -- Comment 3
    variable var1 : integer;   -- Comment 4

  begin

      if something then   -- Comment 5
        a <= b;   -- Comment 6
      end if;  -- Comment 7

  end process;

end architecture;
```

Comments 3 through 7 are all within a process statement, so you would want to look for a `process` rule that would handle alignments of comments.  Referencing the process rules we find two that handle alignment:  `process_034` and `process_035`.

[image]

[image]

Notice that `process_034` will align Comment 3 and Comment 4 only, while `process_035` will alignment comment 5 through 7.

If we wanted to disable aligning of comments 1 and 2, we would look in the architecture rules and find `architecture_027`:

[image]

Conclusion
-------------

So this was a long winded example to tell you that to keep VSG from moving the comment after the `end if` you need to disable rule `process_034`.  My hope is that by explaining the thought behind the classification that it will help you in the future.

Regards,

--Jeremy

sibeov, 2022-02-02:
Thank you for a thorough explanation! Much appreciated!
This solved my "issue". 

Looking forward to future releases and the further development of the tool.

All the best!
-- sibeov

closed 2022-02-02 by sibeov