jeremiah-c-leary/vhdl-style-guide #729: default style misses upper case STD_LOGIC in one case

jancumps, 2022-02-06:
**Environment**
VHDL Style Guide (VSG) version: 3.6.0
Git commit SHA: 9e3799a
Ubuntu 20

**Describe the bug**
I'm using the standard style to validate and fix my code.
It finds all upper case STD_LOGIC and replaces them with std_logic. As expected.
Except this case:
```
  signal busy_prev     : std_logic;
  signal data_to_write : std_logic_vector(7 downto 0);
  signal data_read     : std_logic_vector(7 downto 0);

  procedure i2c_commands (
    init : in STD_LOGIC) is -- if init, the reset command will be added. In that case, mandatory wait 1/32 second
  begin
```

**To Reproduce**
Steps to reproduce the behavior:
write VHDL code like this:
```
procedure i2c_commands (
    init : in STD_LOGIC) is -- if init, the reset command will be added. In that case, mandatory wait 1/32 second
  begin

```
run 
`vsg -f <filename>.vhd --fix`

**Expected behavior**
I was expecting that the script would spot that in this sentence:
`    init : in STD_LOGIC) is -- if init, the reset command will be added. In that case, mandatory wait 1/32 second`
the `STD_LOGIC` was replaced by `std_logic`

jeremiah-c-leary, 2022-02-07:
Hey @jancumps,

Can you see a case in which someone would want a different case for the std_logic depending on where it is in the file?

For example:
* Uppercase in port declarations
* Lowercase in signal assignments

--Jeremy

jancumps, 2022-02-07:
@jeremiah-c-leary ,
_> Can you see a case in which someone would want a different case for the std_logic depending on where it is in the file?_

No. That was not what I was trying to flag.
I had a file with mixed cases for the types (there were `STD_LOGIC` and `std_logic` occurences intermixed in that source file), and the VGG default style nicely flagged - and fixed - almost all the upper case ones.
Except for that one that I was flagging above. It was a type in upper case. VSG did not detect or fix it.

jeremiah-c-leary, 2022-02-09:
Hey @jancumps,

I just pushed an intermediate fix for this issue to the issue-729 branch.  If you get a chance could you check it out on your end and let me know how it works?

I believe I need to deprecate the following rules:
  `signal_010`
  `signal_011`
  `constant_011`
  `variable_010`

As they should be covered by `subtype_002` and `type_014`, as long as they are in the same file.
The ieee types will be handled by the new `ieee_500` rule I created.

jeremiah-c-leary, 2022-02-09:
Hey @jancumps,

I deprecated the rules I noted above.  I am ready to merge to master if you find the update works for you.

--Jeremy

jancumps, 2022-02-09:
@jeremiah-c-leary , I'm going to test.
I installed VSG via PIP. I'll need to build from sources to try out the latest commits.
I'll report back