jeremiah-c-leary/vhdl-style-guide #730: Add rule similar to `library_600` for use clause

staerz, 2022-02-07:
**Is your feature request related to a problem? Please describe.**
The rule [`library_600`](https://vhdl-style-guide.readthedocs.io/en/latest/library_rules.html#library-600) "checks the logical_name in a library_clause has proper case".

This would e.g. pick on the "IEEE" library to be referenced in all capitals.

In some if not many cases, the `library` keyword is followed by a `use` keyword specifying a subset of packages to load.

I couldn't find a rule which would check on e.g.
```vhdl
  use IEEE.math_real.all;
```

**Describe the solution you'd like**
I would like a rule to check the proper casing of the library name (and possibly also the full name of the used package) to follow the global settings.

**Describe alternatives you've considered**
Not caring is not an option. :rofl:

**Additional context**
I hope I overlooked something in the documentation, but the current `master` (3.7.0-5-g1d75e565) of VSG simply ignores the `use` lines it seems.

jeremiah-c-leary, 2022-02-07:
Hey @staerz,

What if we had the following options:
```yaml
rule:
  use_600:
    'library' : 'lower'
    'package' : 'lower'
    'declaration' : 'lower'
    'ieee': 'upper'
    'all': 'lower'
```
That would give you full control over how to format the `selected_name` of a use clause.

--Jeremy

staerz, 2022-02-08:
Hi @jeremiah-c-leary,

that sounds like a good proposal.

I'm wondering if there might be other library names for which a particular casing would be preferred and how that could be implemented by VSG. So yes, "IEEE" is a very popular example (some people arguing that it should always be in all caps, others that it doesn't matter ... as it's VHDL and in the end a style question which is what we have VSG for in the first place), but one might also have some other (e.g. custom) library that one wants to always write "AbcD" or so.
How could that best be matched?

Would it be possible to have something similar to our prefix and suffix solutions that allow you to define a list of predefined exceptions? Instead of the 'ieee' attribute, we'd then have sth. like 'libraries', being a list of predefined library names (with their intended writing). What VSG should then also do is to force any case insensitive match of the library name to the element in that list to that defined way of writing.

Example:
Let's define
```
'libraries': 'IEEE AbcD UVM someTHING'
```

and the code
```vhdl
  use ieee.math_real.all;
...
  use uvM.thingy;
...
  use abcd.thishere.all;
...
  use SOMEthing.else.all;
```
should be forced to
```vhdl
  use IEEE.math_real.all;
...
  use UVM.thingy;
...
  use AbcD.thishere.all;
...
  use someTHING.else.all;
```

Implementing this would by consistency require to extend the rule `library_006` to also be sensitive to the `'libraries'` definition, i.e. that rule should also have that configuration (can 2 rules use the same configuration?)

jeremiah-c-leary, 2022-02-13:
> Implementing this would by consistency require to extend the rule library_006 to also be sensitive to the 'libraries' definition, i.e. that rule should also have that configuration (can 2 rules use the same configuration?

It sounds like you are suggesting we add another entry into a configuration:
```yaml
  library:
     - IEEE
     - UVM
     - AbcD
     - someTHING
```
which could then be used by multiple rules.  These would form the exceptions to casing rules that include libraries.

I think that has merit.  I will think about how it would work as I implement this feature.

--Jeremy

jeremiah-c-leary, 2022-02-14:
Hey @staerz,

This is still a work in progress, but I just pushed my first attempt at this feature to the issue-730 branch.  Could you take a look and let me know what you think?

Thanks,

--Jeremy

staerz, 2022-02-14:
Hi @jeremiah-c-leary,

so I just looked at the diff without having tried running anything.

Your statement "which could then be used by multiple rules" made me think about the following: Couldn't one generalise the effort? In the sense that there are global settings for casing, and each (casing) rule can have its dedicated adjustment. Now would it make sense to have a dedicated setting of exceptions, i.e. a list of words with their intended writing, as a configuration? Each rule could then have its own exceptions (i.e. "IEEE" for the library rule, and "someTHING" for the package rule, etc.

From an implementation point of view that should allow you to have a base class checking for the casing, but applying dedicated rules each time.

jeremiah-c-leary, 2022-02-16:
Hey @staerz,

> From an implementation point of view that should allow you to have a base class checking for the casing, but applying dedicated rules each time.

I seems like this should be possible.  If we define a field in a configuration file:
```yaml
   case_exceptions:
      - IEEE
      - someTHING
```
After all the rule objects have been created, I pass a configuration object to it.  It is the responsibility of the rule to configure itself based on the configuration it is given.  So I could add something to the configuration method to add in the "global" case exceptions.

Someone could then override them with specific exceptions on a rule by rule basis.

Would you like me to implement this?

--Jeremy

staerz, 2022-02-16:
Hi @jeremiah-c-leary,

yes, I think this would be a valuable addition to VSG. The simple question is if one would want that `case_exception` to be global, i.e. to be applied for all rules checking casing, or if one would want to have a `case_exception` dedicated to each rule.

E.g. do we want to be able to force `IEEE` as the library name, but not allow `IEEE` as the name of a procedure for example.

At that point I'm not sure if the one would have a benefit over the other but I tend to having an exception definition per rule.
There may then be overlap between some items, but I hope that would be considerably few items.

So that's my 5 cents on that.

jeremiah-c-leary, 2022-02-17:
Hey @staerz,

> The simple question is if one would want that case_exception to be global, i.e. to be applied for all rules checking casing, or if one would want to have a case_exception dedicated to each rule.

Both options would be available for certain rules:
```yaml
case_exceptions:
  - IEEE
  - someTHING
rule:
  use_clause_500:
     exceptions:
        - IEEE
        - myOtherLibrary
  use_clause_501:
    exceptions:
        - MyPackage
  architecture_011:
    exceptions:
        - MySpecialArchitectureName
  procedure_501:
    exceptions:
        - ieee
```
In this configuration the `case_exceptions` would be applied to all case rules marked with `case::name`.  So all rules that deal with identifiers.

The individual rules would override the `case_exceptions`.  Essentially exceptions to the exceptions.

--Jeremy

staerz, 2022-02-17:
Yes, I could subscribe to that solution! [with the tiny correction that the configuration keyword for each rule should also be `case_exceptions`, right?]

jeremiah-c-leary, 2022-03-27:
Hey @staerz,

I changed the rule configuration to be `case_exceptions` and also added it to `global`:
```yaml
rule:
    global:
        case_exceptions:
           - IEEE
           - someTHING
    architecture_013:
        case : lower
    architecture_011:
        case_exceptions:
           - MySpecialArchitectureName
```

One question though....I currently check for a whole word to apply the exception to.  So in the case of architecture_013 with this  snippet:
```vhdl
architecture my_IEEE_alternative of fifo is
```
would still fail and be fixed to:
```vhdl
architecture my_ieee_alternative of fifo is
```

Does this match your expectations?

--Jeremy

jeremiah-c-leary, 2022-03-27:
Hey @staerz,

I have pushed an update to the issue-730 branch.  When you get chance let me know what you think.

--Jeremy

staerz, 2022-03-30:
Hi @jeremiah-c-leary,

yes, I would say a full word match is what should be done, so yes, this example with the architecture name would (and should) fail, I'd say.

Maybe add as a comment into your source code that this could, based on user feedback, be changed to a more general match (within words).

I tried your new version and made VSG crash :rofl:

I updated to your latest commit on `issue-730` and then added this to our config:
```json
    "global":{
      "case_exceptions":[
        "IEEE",
        "PoC",
      ],
    },
```

Yes, the first files get reported correctly to change e.g. `poc` to `PoC` and `ieee` to `IEEE` now, but also I get:

```
================================================================================
File:  components/fpga/context/registers.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 424
Total Violations:    6
  Error   :     6
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  function_012              | Error      |        148 | Move : -6 columns
  subprogram_body_400       | Error      |        153 | Move := -1 columns
  subprogram_body_400       | Error      |        155 | Move := -1 columns
  function_012              | Error      |        200 | Move : -6 columns
  function_012              | Error      |        202 | Move : -5 columns
  subprogram_body_400       | Error      |        207 | Move := -5 columns
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
while there was no complaint about that file earlier (and I haven't touched it).

And I also get this:
```
================================================================================
File:  components/fpga/context/mapping.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 586
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  library_600               | Error      |         16 | Change "ieee" to "IEEE"
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/apply_rules.py", line 98, in apply_rules
    oRules.check_rules(
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rule_list.py", line 223, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rule.py", line 139, in analyze
    self._analyze(lToi)
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rules/token_case.py", line 44, in _analyze
    oViolation = case_utils.check_for_case_violation(oToi, self, check_prefix, check_suffix, check_whole)
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rules/case_utils.py", line 35, in check_for_case_violation
    oViolation = check_for_lower_case_with_whole_exception(sObjectValue, self, oToi, iIndex, iMyLine)
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rules/case_utils.py", line 181, in check_for_lower_case_with_whole_exception
    if exception_found(sObjectValue, self):
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rules/case_utils.py", line 188, in exception_found
    if sObjectValue.lower() in self.case_exceptions_lower:
AttributeError: 'rule_506' object has no attribute 'case_exceptions_lower'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==3.0.0.dev20', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/__main__.py", line 138, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 868, in next
    raise value
AttributeError: 'rule_506' object has no attribute 'case_exceptions_lower'
```

Let me know if you think I made a mistake in a configuration, but to me it looks like you forgot about 1 rule or so (which is fully acceptable given the complexity :wink: )

jeremiah-c-leary, 2022-04-02:
Hey @staerz,

I tracked down the issue with the crash and pushed an update to the branch.

Still looking at the new violations which cropped.  It seems odd to induce alignment errors when adjusting for case though.

--Jeremy

staerz, 2022-04-04:
Hi @jeremiah-c-leary,

alright, thanks for the update.

I just ran it on the very same code base and we got one step further, but the similar error still exists for another rule it seems:
```
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/apply_rules.py", line 98, in apply_rules
    oRules.check_rules(
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rule_list.py", line 223, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rule.py", line 139, in analyze
    self._analyze(lToi)
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rules/token_case.py", line 44, in _analyze
    oViolation = case_utils.check_for_case_violation(oToi, self, check_prefix, check_suffix, check_whole)
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rules/case_utils.py", line 35, in check_for_case_violation
    oViolation = check_for_lower_case_with_whole_exception(sObjectValue, self, oToi, iIndex, iMyLine)
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rules/case_utils.py", line 181, in check_for_lower_case_with_whole_exception
    if exception_found(sObjectValue, self):
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/rules/case_utils.py", line 188, in exception_found
    if sObjectValue.lower() in self.case_exceptions_lower:
AttributeError: 'rule_010' object has no attribute 'case_exceptions_lower'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==3.0.0.dev20', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/dist-packages/vsg-3.0.0.dev20-py3.8.egg/vsg/__main__.py", line 138, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 868, in next
    raise value
AttributeError: 'rule_010' object has no attribute 'case_exceptions_lower'
```

I'd say don't worry for the indentation of the colon and stuff for now, let's try to focus purely on the new functionality and then we can see to track that down. I'm sure we'll get there eventually.

staerz, 2022-04-04:
Hi @jeremiah-c-leary,

note that I saw the indentation/alignment complaints also showing up when testing branch issue-539, so really don't worry. I think it is due to the fact that I switch between VSG 3.0.0 and what ever branch you point me to.
So there may be a few changes in between for which I'm not yet fully prepared (or some bugs have smuggled in, but in that case, don't worry, I'd open issues at the given time once I know more if it's me or VSG :wink: ).

jeremiah-c-leary, 2022-04-12:
Hey @staerz,

I am unable to reproduce the current crash you listed.  Could you give me an example that fails?

Thanks,

--Jeremy

staerz, 2022-04-13:
Alright, I found the file which it has troubles with:
[lolli_xgbe_top.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/8484304/lolli_xgbe_top.txt)

(As usual, you'll have to rename it to `*.vhd`)

I'm not saying that this is the only file it would crash on, but it's the first one that it encounters and I also reduced my local config temporarily to only check that file and it fails with the exact same error message.

I hope that you can make sense of it now.

jeremiah-c-leary, 2022-04-14:
Hey @staerz,

I was able to replicate the failure!!!  Working on a solution now.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-04-14:
Hey @staerz,

I pushed  a fix for the latest error.  Your example file does not crash anymore.  Please let me know if there are additional problems on the branch.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-04-15:
Hey @staerz,

I was not processing the selected_name for context references the same as use clauses.

I have pushed an update to the branch.  I tested your example and it appears to work.

I did deprecate the rule `context_004` and created rules `context_500` and `context_501`.

Let me know what you think.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-04-19:
Woot!!!

I will get it merged in.

PR #741 (2022-04-20): Updates for issue #730
Adding rules to handle case for use_clause selected names.

  1)  Added new tokens to use_clause
  2)  selected names now parsed with new tokens
  3)  Added rules:
      a)  use_clause_500
      b)  use_clause_501
      c)  use_clause_502
      d)  use_clause_503
  4)  Added tests

Resolves #730
commit 267b17e7 (2022-04-20): Updates for issue #730 (#741)

* Updates for issue #730

Adding rules to handle case for use_clause selected names.

  1)  Added new tokens to use_clause
  2)  selected names now parsed with new tokens
  3)  Added rules:
      a)  use_clause_500
      b)  use_clause_501
      c)  use_clause_502
      d)  use_clause_503
  4)  Added tests

Resolves #730

* Refactoring to reduce duplication.
Removing trailing whitespace.

* Changing exceptions to case_exceptions.

1)  case::name rules allow case_exceptions by default
2)  Updated instantiated_unit to split entity_name with a library name
3)  Rule  [... cut]
closed 2022-04-20 by jeremiah-c-leary