jeremiah-c-leary/vhdl-style-guide #742: `comment_010` proposes putting comments unintended indentation in partcular cases

staerz, 2022-02-14:
**Environment**
VSG 3.8.0-2-g9b46a997

**Describe the bug**
`comment_010` is put off in the library and case section and wants the comment to be indented one level more than I would naturally put it.

E.g. in the library section, which foresees 1 level of indent for the `use` case, it seems that level of indent is also applied for the comment:

```diff
   );
 end entity delay_chain;
 
+--! @cond
 library altera;
   use altera.altera_syn_attributes.all;
+  --! @endcond

 --! Implementation of the #delay_chain
 architecture behavioral of delay_chain is
```

This example just attempts to put the library and use clause in an exclusion clause for Doxygen. Note that Doxygen is super sensitive to blank lines as that's what makes Doxygen separate the individual comments and code elements, so there's no room for inserting another blank line or removing one here.

Note that the `endcond` is indented 1 more level than expected (it is not made a comment related to the `use` keyword, but to the higher level hierarchical element (`library`) here. It is though understood that the next line could potentially hold another `use` keyword which would explain the requested indent. It looks like a "look ahead" functionality is missing here.

In a case section a similar behaviour can be observed:
```diff
 --! Implementation of the #delay_chain
 architecture behavioral of delay_chain is
@@ -109,10 +111,11 @@ begin
   -- now set the last item in the chain
   --! @cond
   gen_delay : case D_DEPTH generate
-    -- No delay of #sig_i
+      -- No delay of #sig_i
     when 0 =>
       delay_reg(0) <= sig_i;
-    -- Delay #sig_i by one clock cycle
+      -- Delay #sig_i by one clock cycle
     when 1 =>
```

Note that this time the diff shows actual modifications that VSG imposed after having enabled the `comment_010` rule.

Here, also the indent is one more level than expected. Similarly, it seems that VSG tries to relate the comment to the line before the line containing the comment, but not the line after (which may have different indentation level).

**To Reproduce**
Steps to reproduce the behavior:
1. Have code similar to the reported one
2. Run VSG (in default configuration) on the above code
3. Observe the error

**Expected behavior**
In the instances reported, I would expect VSG to require the comment with one level of indent less.

**Additional context**
Relates to #542.

jeremiah-c-leary, 2022-04-16:
Hey @staerz,

I pushed an update for this to the issue-742 branch.  When you get a chance, could you check it out.

Thanks,

--Jeremy

staerz, 2022-04-19:
Hi @jeremiah-c-leary,

I confirm that the provided solution on branch 742 fixes the reported issue.

Now I looked a bit more and there are further examples where `comment_010` kicks in as it's being put off by code parsing and a missing "look ahead".

Here are the examples I found in our code base:
```vhdl
    gen_feb2_11 : if G_FEB2_LINK_LOLLI_N <= 11 generate
      -- Use bank 4D for first 6 links and neighboring 4E for next 5 links
      feb2_rx(i) <= feb2_rx_p(i + 2);
    elsif G_FEB2_LINK_LOLLI_N <= 22 generate
      -- Use neighboring banks 4K, 4L, 4M, 4N to fit 22 links
      feb2_rx(i) <= feb2_rx_p(i + 20);
    -- the next step is a not fully populated tile and a full tile
    elsif G_FEB2_LINK_LOLLI_N <= 84 generate
```
The last comment is complained on ("Indent level 3") as VSG thinks it should be
```vhdl
      feb2_rx(i) <= feb2_rx_p(i + 20);
      -- the next step is a not fully populated tile and a full tile
    elsif G_FEB2_LINK_LOLLI_N <= 84 generate
```
(but of course, I don't want that...)

The second is sth. that I thought we had already solved and I looked it up and we actually already [discussed it here](https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/542#issuecomment-1041871202):

In a `package body` I have that:
```vhdl
  end function reg_write;

-- vsg_off comment_010
-- todo: need to figure out if somehow the ambiguity between unsigned and std_logic_vector could be fiddled out
--  function reg_write (addr: std_logic_vector; data: unsigned) return t_avmm_master is
--  begin
--  end function reg_write;
--
--  function reg_write (addr: unsigned; data: std_logic_vector) return t_avmm_master is
--  begin
--  end function reg_write;
-- vsg_on comment_010

  function reg_write (addr: std_logic_vector; data: std_logic_vector) return t_avmm_master is
```
and VSG complains about the line
```vhdl
-- vsg_on comment_010
```

Would you prefer solving these 2 particular cases within this issue or should I rather open a new one (note that the title of this issue is vague enough to call in for this extension, but I agree, it's not part of the initial issue description).

jeremiah-c-leary, 2022-04-20:
Hey @staerz,

> Now I looked a bit more and there are further examples where comment_010 kicks in as it's being put off by code parsing and a missing "look ahead".

I believe I need to re-think how I am setting the indent of comments.

My current method is to assign the comment to the "current" indent level and then use rules like `case_021` to adjust the indent.

I should switch all comments to be indented to the next indent.

I will think on how to do this.

> The second is sth. that I thought we had already solved and I looked it up and we actually already https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/542#issuecomment-1041871202:

I thought we fixed that issue already.

> Would you prefer solving these 2 particular cases within this issue or should I rather open a new one (note that the title of this issue is vague enough to call in for this extension, but I agree, it's not part of the initial issue description).

Let's keep it within this issue for now.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-04-20:
> I should switch all comments to be indented to the next indent.

Turns out it is a little more complicated than  that, but I do have a "look ahead" function working.  I just need to tweak the indent value to use.

--Jeremy

jeremiah-c-leary, 2022-04-20:
Hey @staerz,

I suppose it comes down to how one would want the following comment indented:
```vhdl
architecture rtl of fifo is
  -- Comment
begin
  -- Comment
end architecture;
```
Should the comment be indented or moved to the right?
```vhdl
architecture rtl of fifo is
-- Comment
begin
-- Comment
end architecture;
```
There are many cases like this:
```vhdl
package fifo_pkg is
-- Comment
end package;

package body fifo_pkg is
-- Comment
end package body;

process 
-- Comment
begin
-- Comment
end process;
```
For some reason these seem different than the following:
```vhdl
if a = 1 then
-- Comment
elsif a = 2 then
-- Comment
else a = 3 then
  -- Comment
end if;
```
maybe because the `begin` and `end` keywords do not really need any explanation?

Any thoughts on this?

--Jeremy

staerz, 2022-04-20:
Hi @jeremiah-c-leary,

so I'd say the comment should **always** match the indentation of the following keyword's indentation (**always** look ahead).
For me, the question "how one would want the following comment indented" doesn't even pose itself.

That is, in all your cases, the comment should be all the way to the left.

Only if more code is added that then populates these structures, indented comments would apply.

I.e. your
```vhdl
architecture rtl of fifo is
  -- Comment
begin
  -- Comment
end architecture;
```
would apply **only if** the complete picture would be
```vhdl
architecture rtl of fifo is
  -- Comment (regarding the signal abc)
  signal abc : std_logic;
-- Comment (regarding the begin of the architecture)
begin
  -- Comment (regarding the assignment of abc)
  abc <= something;
-- Comment (regarding the end of the architecture)
end architecture;
```

Now, of course, I do not expect many people to add comments to their `begin` and `end` keywords of structures, but if they wished, that's how I would expect the indentation to be.
And note that this would certainly apply to structures that go beyond the simple `begin` and `end`, i.e. all structures involving forks, i.e. `if` (esp. the `else` and `elsif` path) or `case` (esp. the `when` keyword): There are comments intended for the code inside that structure (which should follow the additional level of indent) and those commenting the fork structure itself (which should follow the indent of these, respectively) as I gave in my example above.

staerz, 2022-04-21:
Hi @jeremiah-c-leary,

I have tested it and your "look forward" works like a charm!

Now we're only left with the complaint on
```vhdl
-- vsg_on comment_010
```

jeremiah-c-leary, 2022-04-22:
Hey @staerz,

> I have tested it and your "look forward" works like a charm!

That is awesome!!

> Now we're only left with the complaint on

While tracking this issue down I ended up rewriting how the code tags were applied.  I'm not sure what I was thinking with how it was implemented before.

I pushed an update to the issue-742 branch.  When you get a chance, could you validate it?

Thanks,

--Jeremy

staerz, 2022-04-22:
Hi @jeremiah-c-leary,

> I'm not sure what I was thinking with how it was implemented before.

:rofl: Yeah, that happens to me sometimes, too. But you know how it is, you always end up going to bed a little bit more knowledgeable than you woke up that day.

So I gave it a test and am delighted to report that now this is sooper dooper all working!

Make that issue closed by getting the merge request merged :+1: 

And let me know what's next on your todo list :wink:
If I could make a wish, it would be #549 (as it has a template solution already as of #499) [which now actually seems related to #539 and we might be on a more global track here as that doesn't only apply to signals and variables but also to constants]

jeremiah-c-leary, 2022-04-23:
Hey @staerz,

I am glad to finally get this issue closed!!!

> you always end up going to bed a little bit more knowledgeable than you woke up that day.

Complete agree.  :smile:

> And let me know what's next on your todo list

My current plan is to finish #539, which should then allow me to address these issues:  #549, #677 and #699.

I really appreciate you finding issues and helping me close them.

Regards,

--Jeremy

closed 2022-04-23 by jeremiah-c-leary

staerz, 2022-04-25:
Hi @jeremiah-c-leary,

I'm happy to hear you feel that way. But finding issues and complaining about them is not as much of a job as actually fixing it like you do.

So overall I think it takes both of us :smiley: 

I'm happy to hear your plan as it nicely aligns with my current desired items.

PR #754 (2022-04-23): Updates for issue #742:
Adjusting indenting for comments near use_clauses and case generates.

  1)  Added rule case_generate_alternative_300
  2)  Changed indenting of comments after use_clauses

Resolves #742