jeremiah-c-leary/vhdl-style-guide #747: Rule conflict similar to #479

imd1, 2022-03-19:
**Environment**
VSG 3.9.0, Centos 7

**Describe the bug**

Here's the code snippet:

```

architecture rtl of an_architecture is

begin

  PROC_XXX : process (CLOCK, RESET) is
  begin

    if RESET = '1' then

      for I in 0 to G_NUM_SPI_REGS_RD - 1 loop

        MUX_SPI_REGS_RD(I) <= (others => '0');

      end loop;

    elsif rising_edge(CLOCK) then
      adc_done_d1 <= adc_done;
    end if;

  end process PROC_XXX;

end architecture rtl;
```

When VSG is run with default settings, there are unfixable errors...

```
================================================================================
File:  /path/to/file.vhd
================================================================================
Phase 3 of 7... Reporting
Total Rules Checked: 310
Total Violations:    2
  Error   :     2
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  if_006                    | Error      |          9 | Remove blank lines below
  if_007                    | Error      |         17 | Remove blank line(s) before the *elsif* keyword.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

```

Seems like if_006 is conflicting with loop_statement_200 (?) and if_007 with loop_statement_203 (?)

jeremiah-c-leary, 2022-03-26:
Hey @imd1,

Yeah, I could see something like that happening.

I like the blank lines before and after the loop though.  It really makes it stand out.  Maybe I could put some exceptions in the blank line check for if_006 and if_007 for lines not being a sequential signal assignment.

--Jeremy

jeremiah-c-leary, 2022-03-28:
Hey @imd1,

Turns out I had already put in some code to allow for case statements so I just had to add in the loop_statement tokens.  I pushed a fix for this to the issue-747 branch.  When you get a chance could you check it out.

I will plan to push this to master by 04/01 if I do not hear from you.

Thanks,

--Jeremy

PR #749 (2022-03-31): Updates for issue #747
Resolving conflict between if rules and loop_statement rules regarding blank lines.

Adding overrides to if blank line rules to allow loop_statements to be one blank line away.

Resolves #747
closed 2022-03-31 by jeremiah-c-leary