jeremiah-c-leary/vhdl-style-guide #750: Real numbers and code tag

imd1, 2022-03-31:
**Environment**
VSG 3.9.0, Window 10, Python 3.10.0

**Describe the bug**
VSG incorrectly mistakes exponential real numbers with negative numbers as the difference between two numbers. When a code tag is added to disable all VSG rules to avoid this problem, there is no effect. Have to specify the offending rule.

Happens in a package construct

**Screenshots**

The following results in no change

```
  -- vsg_off whitespace_011 : turn off vsg to avoid e notation corruption
  constant WIN_FLT_C : win_flt_t := (
    3.55149375e-04, 9.57857540e-03, 4.97092372e-02, 1.51849128e-01,
    3.35404663e-01, 5.81810162e-01, 8.25681302e-01, 9.79118935e-01,
    9.79118935e-01, 8.25681302e-01, 5.81810162e-01, 3.35404663e-01,
    1.51849128e-01, 4.97092372e-02, 9.57857540e-03, 3.55149375e-04
  );
  -- vsg_on
```

When the rule code tag is removed, VSG corrects to this:

```
  -- vsg_off : turn off vsg to avoid e notation corruption
  constant WIN_FLT_C : win_flt_t := (
    3.55149375e - 04, 9.57857540e - 03, 4.97092372e - 02, 1.51849128e - 01,
    3.35404663e - 01, 5.81810162e - 01, 8.25681302e - 01, 9.79118935e - 01,
    9.79118935e - 01, 8.25681302e - 01, 5.81810162e - 01, 3.35404663e - 01,
    1.51849128e - 01, 4.97092372e - 02, 9.57857540e - 03, 3.55149375e - 04
  );
  -- vsg_on
```

So this highlights 2 problems
* why did the code tag not work?
* why is the real number being corrupted?

jeremiah-c-leary, 2022-03-31:
Hey @imd1,

The code tag worked in my snippet:
```vhdl
  1
  2 package issue_750_pkg is
  3
  4   -- vsg_off whitespace_011 : blah
  5   constant my_con : blah :=
  6   (
  7     4.5e-10,
  8     6.1e-11
  9   );
 10   -- vsg_on
 11
 12 end package issue_750_pkg;
```
So I am not sure why it did not work for you.

The real number is being corrupted because the `-` is being classified as a `minus` instead of a `sign`.

I can work on the second issue and we can investigate the first one.

--Jeremy

imd1, 2022-03-31:
> 

I had expected the `vsg_off` without the whitespace rule would have turned off ALL rules, which it did not do.

jeremiah-c-leary, 2022-04-02:
I would have expected it to work also.  I updated my code snippet to this:
```vhdl
  1
  2 package issue_750_pkg is
  3
  4   -- vsg_off whitespace_011 : blah
  5   constant my_con : blah := (
  6      4.5e-10,
  7      6.1e-11
  8   );
  9   -- vsg_on
 10
 11   -- vsg_off
 12   constant my_con : blah := (
 13      4.5e-10,
 14      6.1e-11
 15   );
 16   -- vsg_on
 17
 18   constant my_con : blah := (
 19      4.5e-10,
 20      6.1e-11
 21   );
 22
 23 end package issue_750_pkg;
```
...and when I run `--fix` with the base rules I get:
```vhdl
  1
  2 package issue_750_pkg is
  3
  4   -- vsg_off whitespace_011 : blah
  5   constant my_con : blah :=
  6   (
  7     4.5e-10,
  8     6.1e-11
  9   );
 10   -- vsg_on
 11
 12   -- vsg_off
 13   constant my_con : blah := (
 14      4.5e-10,
 15      6.1e-11
 16   );
 17   -- vsg_on
 18
 19   constant my_con : blah :=
 20   (
 21     4.5e - 10,
 22     6.1e - 11
 23   );
 24
 25 end package issue_750_pkg;
```
You can see how the real numbers are not messed up in the first two constant declarations, but is in the last.  So it appears the code tags are working in my example.

Could you run my snippet above with your configuration?  I am curious if there is a previous code construct that somehow negates the code tag in your original file.

Meanwhile, I will work on the real number issue.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-04-02:
Hey @imd1,

I just figured it out.  I was not expecting anything after the `--vsg_off`.

I like you idea to use a colon to separate the rules from the comment, so I update the code tag logic to handle the colon correctly.

I pushed an update for that issue to the issue-750 branch and currently working the real number part.

--Jeremy

jeremiah-c-leary, 2022-04-02:
Hey @imd1,

I believe I fixed the parsing of real numbers.  I still need to do a bit of refactoring, but if you could check out what I have on the issue-750 branch I would appreciate it.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-04-03:
Hey @imd1,

I finished my refactoring so if the updates work for you I am ready to merge it in.

I also added rule `exponent_500` to set the case of the `e`.

I will plan to merge this in by 4/8 for a release next weekend.

Thanks,

--Jeremy

closed 2022-04-11 by jeremiah-c-leary