jeremiah-c-leary/vhdl-style-guide #755: signal_014, constant_013 and architecture_600 don't work properly when more than one entity is declared in a file.

Timandrias, 2022-04-22:
**Environment**
Launching VSG from the command line.

**Describe the bug**
If in one file there are several entities, and in one of them there are signals (or constants) with the same name than ports of other entity, VSG does not bound _signal_014_ (or _constant_013_) to the relevant entity where the signal was declared, but extends it to the others, saying that the ports in other entities have wrong capitalization.

The rule _architecture_600_ is behaving in a similar way, assuming that constants with the same name of generics of another entity are generics. 

**To Reproduce**
Run a simple check with the attached config file to the attached vhdl file.
vsg -c .\vsg_config.json -f .\minimal_bug.vhd
[minimal_bug.zip](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/8539577/minimal_bug.zip)

**Expected behavior**
I would expect these rules to be bounded to one entity.

jeremiah-c-leary, 2022-04-22:
Hey @Timandrias,

This is a known limitation of VSG.  It currently does not have the data structures necessary to properly enforce consistent case rules except in limited cases.

For example, if the entity and architecture are split into separate files:
entity file:
```vhdl
entity FIFO is
  generic (
     G_WIDTH : natural := 16
  );
end entity FIFO;
```
architecture file:
```vhdl
architecture RTL of FIFO is

  signal w_data : std_logic_vector(g_width - 1 downto 0);

begin
```
then `architecture_600` would not work as VSG would not have access to the entity file when the architecture file is being processed.

I think I have reached the limit of my current data structure and need to upgrade it to hand this, and other, limitations.

So unfortunately I do not have a quick solution to this issue, but it is in my road map.

Regards,

--Jeremy

Timandrias, 2022-04-22:
That's unfortunate. Maybe it would be a good idea to add a section in the docs mentioning these kind of "nofix until rewrite" issues.  I tried to search in the github issues to see if there was something similar, but I couldn't find anything... 

Thank you, anyway :)

jeremiah-c-leary, 2023-11-25:
Morning @Timandrias ,

Just wanted to give you an update on this issue.  I am currently working on a fix for `variable_011` which should also apply to these rules.  I am hoping to have an update in the next week.

--Jeremy

jeremiah-c-leary, 2023-11-25:
Good Afternoon @Timandrias .

I have pushed an update for this issue to the `issue-551` branch.  I still have to fix `architecture_600` and `entity_600`, but the other consistent case rules have been updated.  When you get a chance could you try this out on your end and let me know if there are any issues?

I will start working on the last two rules tomorrow.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-11-26:
Morning @Timandrias ,

I have updated rules `ardchitecture_600` and `entity_600` to handle multiple instances in a single file.  Let me know how it works out for you.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-12-09:
Morning @Timandrias ,

Just wanted to ping you on this issue to see if you had time to check it out.

Thanks,

--Jeremy

jeremiah-c-leary, 2024-01-21:
Afternoon @Timandrias ,

I just wanted to let you know that I will be merging to this master and closing the issue.  If you find additional problem, please create another issue.

Regards,

--Jeremy

PR #1052 (2024-01-21): Issue#551:  Fixing scoping of consistent token case rules.
Resolves #551
Resolves #755
Resolves #998 
Resolves #1023 
Resolves #1062