jeremiah-c-leary/vhdl-style-guide #758: Documentation mistake

hcommin, 2022-04-25:
The "fix" code snippet for `assert_005` is incorrect:
https://vhdl-style-guide.readthedocs.io/en/latest/assert_rules.html#assert-005

 It should put `severity` on its own line (similar to `assert_004`). And it shouldn't put `report` on its own line.

### Current state:
```vhdl
architecture rtl of fifo is
begin

  process begin

    assert WIDTH > 16
      report "FIFO width is limited to 16 bits." severity FAILURE;

  end process;

end architecture rtl;
```

### Should be:
```vhdl
architecture rtl of fifo is
begin

  process begin

    assert WIDTH > 16 report "FIFO width is limited to 16 bits."
      severity FAILURE;

  end process;

end architecture rtl;
```

jeremiah-c-leary, 2022-04-26:
Hey @hcommin,

Thanks for finding those issues in the documentation.  I have pushed an update that appears to fix those issues.

If you have time, could validate the changes were made.

I will plan to close this issue by Apr 29th if I do not hear back from you by then.

Regards,

--Jeremy

hcommin, 2022-04-26:
@jeremiah-c-leary Your push fixes them all (according to my understanding). The quality of the documentation is excellent. Thank you for this fantastic tool.

closed 2022-04-26 by hcommin