jeremiah-c-leary/vhdl-style-guide #760: Integration with Notepad++ via PythonScript

hcommin, 2022-04-27:
The [PythonScript](https://github.com/bruderstein/PythonScript) plugin for Notepad++ allows the contents of the Notepad++ text editor to be manipulated dynamically using Python 2 or Python 3. It provides various [functions](http://npppythonscript.sourceforge.net/docs/latest/scintilla.html) for doing things like dynamically getting/setting the editor text. These function calls behave as if the user did them manually - so, for example, we can undo/redo each action.

VSG usually operates as a command-line tool, modifying text files directly. It can be invoked this way from Notepad++, but then the text file must be reloaded from disk, which can be inconvenient (particularly because it kills the undo/redo stack).

Therefore, I would like to be able to import `vsg` as a Python package (see also #612), pass it a string (the contents of a VHDL file) and receive the "fixed" output as a string. Apart from passing data as strings, an interface similar to the command-line interface would be sufficient. This would allow me to implement something like a VHDL "beautify" function in Notepad++:

```python
import vsg

# Read text from Notepad++ editor
text = editor.getText()

# Apply VSG fixes
text = vsg.fix(text, "C:/path/to/config.yaml")

# Write back to Notepad++ editor
editor.setText(text)
```

I would be open to other options for integrating with Notepad++ (e.g. a standalone plugin). I just have some experience with PythonScript, and no experience with writing Notepad++ plugins.

jeremiah-c-leary, 2022-04-28:
Hey @hcommin,

I would like to provide a different entry point into VSG for this effort.  I was also thinking of passing objects into and out of VSG through this entry point.  So something like this:

```python
import vsg

# Request a input data object
inputData = vsg.interface.input_data()

inputData.set_text(editor.getText())
inputData.add_configuration("C:/path/to/config.yaml")
# or even support multiple configurations
inputData.add_configuration("C:/path/to/other/config.yaml")
# Then we can add any number of other options
inputData.set_style("indent_only")

# Pass the data object back to the PythonScript interface and get a result object back:
resultObject = vsg.pythonScript.fix(inputData)

# You could check for errors and conditionally update the text
if not result.Object.error:
   editor.setText(resultObject.get_text())
```

What do think?

--Jeremy

hcommin, 2022-04-28:
@jeremiah-c-leary This would meet and exceed my requirements.

I can get by with the command-line interface (and for some of my tasks, it is clearly the way to go). But for things like code reviews and refactoring legacy code, it would really save me a lot of time to be able to integrate properly with Notepad++.

I am currently refactoring a library of ~170 VHDL files, so I very much look forward to this improvement.

hcommin, 2022-05-03:
@jeremiah-c-leary As far as I can see, this is working perfectly. It's actually pretty amazing - from trash to beauty in one keyboard shortcut.

I just have one question: Is there some way I can access the normal command-line output summary and print it to the Python console? I mean this summary:

```
================================================================================
File:  ./test.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 459
Total Violations:    1
  Error   :     0
  Warning :     1
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  variable_600              | Warning    |        182 | Suffix var with one of the following: _v
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

-------------------

Just in case anyone else stumbles across this, here are some usage details...

The only potential setup step is to make sure VSG is included in PythonScript's Python path. For a normal pip install of VSG, this is probably working by default. But for a Git clone (or any kind of unusual Python installation, such as multiple versions installed on the same system):

- Add VSG to PythonScript path:
    - Open: `C:/Program Files/Notepad++/plugins/PythonScript/scripts/startup.py`
        - For VSG Git clone: `sys.path.insert(1, "C:/path/to/vsg/vhdl-style-guide")`
        - For VSG pip install: `sys.path.insert(1, "C:/path/to/Python3/Lib/site-packages")`

I don't use a VSG style because I have set every rule manually. So, in my case, I am using this simple script:

```python
import vsg.interfaces
from vsg.interfaces import notepad_pp as npp

# Create a VSG interface to Notepad++
vsg_interface = npp.New()

# Set up VSG input parameters
vsg_input_params = vsg_interface.get_input_arguments()
vsg_input_params.add_configuration("C:/path/to/workplace_config.yaml")
vsg_input_params.add_configuration("C:/path/to/personal_config.yaml")
vsg_input_params.set_text(editor.getText())

# Fix the text
fixed_text = vsg_interface.fix(vsg_input_params)

# Write back to Notepad++
editor.setText(fixed_text.get_text())
```

jeremiah-c-leary, 2022-05-03:
Hey @hcommin,

> As far as I can see, this is working perfectly.

Awesome.

> It's actually pretty amazing - from trash to beauty in one keyboard shortcut.

Yeah, integrating into the editor really makes coding easier.

It also opens up another use case that I call lazy formatting.  Essentially you just write the code without caring about formatting.  So no more column alignment, no more spacing etc...

I found I could concentrate on getting the idea down easier.  While I paused to think about what to do next, I would hit a macro key and the code would be formatted correctly.

> I just have one question: Is there some way I can access the normal command-line output summary and print it to the Python console? I mean this summary:

I can provide that in the results object.

> Just in case anyone else stumbles across this, here are some usage details...

I plan to add instructions to the [editor integration](https://vhdl-style-guide.readthedocs.io/en/latest/editor_integration.html) section so any information you can provide would be helpful.

--Jeremy

hcommin, 2022-05-04:
@jeremiah-c-leary It took me a minute to realize that `get_stdout()` gives me the text to print. Somehow, I was expecting that I needed to pass some object to `set_stdout()` to tell VSG where to send its output. For example, PythonScript tells Python to output to the Notepad++ console like this:

```python
sys.stdout = console
```

However, once I figured this out, adding this line worked perfectly:
```python
    # Print VSG output to Python console
    print(fixed_text.get_stdout())
```

> I plan to add instructions to the [editor integration](https://vhdl-style-guide.readthedocs.io/en/latest/editor_integration.html) section so any information you can provide would be helpful.

For most users, I would first recommend something equivalent to your Vim example, where the file is modified directly. This way, the file must be reloaded from disk, but there is no need to set up PythonScript (which for Python3 is quite cumbersome). Here are the steps:

1. Open `%APPDATA%/Notepad++/shortcuts.xml` **not** in Notepad++ (e.g. in Vim, Emacs or Windows Notepad).
2. In the `<UserDefinedCommands>` section, add this line:
 ```xml
<Command name="Run VSG" Ctrl="no" Alt="no" Shift="no" Key="120">vsg -f &quot;$(FULL_CURRENT_PATH)&quot; --fix</Command>
```

Pressing F9 will then run VSG on the current file. It's a bit clunky - a cmd window pops up, then disappears, then Notepad++ prompts you to reload the file from disk. But it does work.

For PythonScript, I think it would be sufficient to just provide an example script like this (tidied up according to your preferences):

```python
import vsg.interfaces
from vsg.interfaces import notepad_pp as npp

# Create a VSG interface to Notepad++
vsg_interface = npp.New()

# Set up VSG input parameters
vsg_input_params = vsg_interface.get_input_arguments()
vsg_input_params.add_configuration("C:/path/to/workplace_config.yaml")
vsg_input_params.add_configuration("C:/path/to/personal_config.yaml")
vsg_input_params.set_text(editor.getText())

# Fix the text
fixed_text = vsg_interface.fix(vsg_input_params)

# Print VSG output to Python console
print(fixed_text.get_stdout())

# Write back to Notepad++
editor.setText(fixed_text.get_text())
```

This can be run like any other PythonScript script.

PR #774 (2022-05-05): Issue 760
Resolves #760
closed 2022-05-04 by hcommin