jeremiah-c-leary/vhdl-style-guide #761: Error with nested blocks

hcommin, 2022-04-27:
The [block_007](https://vhdl-style-guide.readthedocs.io/en/latest/block_rules.html#block-007) rule doesn't seem to work correctly with nested blocks.

Minimal example (compiles OK in Modelsim 2020.1):
```vhdl
entity e is
end entity;

architecture rtl of e is
begin
    b_0 : block
    begin

        b_1 : block
        begin

        end block b_1;

    end block;
end architecture;
```

VSG (with default config) reports an unfixable error:

```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  block_007                 | Error      |         16 | Add label
----------------------------+------------+------------+--------------------------------------
```

If I delete the inner block, then no error is reported:
```vhdl
entity e is
end entity;

architecture rtl of e is
begin
    b_0 : block
    begin

    end block;
end architecture;
```

(It looks like GitHub's VHDL syntax highlighting is also having trouble with the nested blocks).

jeremiah-c-leary, 2022-04-28:
Hey @hcommin,

I just pushed a fix for this issue to the issue-761 branch.  When you get a chance could you validate it works on your end.

I will plan to merge this to master on May 6th if I do not hear back from you.

Thanks,

--Jeremy

hcommin, 2022-04-28:
Fixed in my test cases - many thanks. Closing issue.

PR #765 (2022-04-28): Issue 761
Resolves #761
closed 2022-04-28 by hcommin