jeremiah-c-leary/vhdl-style-guide #762: Error with nested generates

hcommin, 2022-04-27:
This issue may be related to #761.

The [generate_403](https://vhdl-style-guide.readthedocs.io/en/latest/generate_rules.html#generate-403) rule doesn't seem to work correctly with nested if-generates.

Minimal example (compiles OK in Modelsim 2020.1):
```vhdl
entity e is
end entity e;

architecture rtl of e is
begin

    g_0 : if true generate
        g_1 : if true generate
            signal Sig0 : bit;
            signal Sig00 : bit;
        begin

        end generate g_1;
    end generate g_0;

end architecture rtl;
```

VSG (with default config) reports an unfixable error:
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  generate_403              | Error      |         11 | Move : -2 columns
  generate_403              | Error      |         11 | Move : -2 columns
  generate_403              | Error      |         11 | Move : -2 columns
----------------------------+------------+------------+--------------------------------------
```

If I remove the outer generate, then no error is reported:

```vhdl
entity e is
end entity e;

architecture rtl of e is
begin

    g_1 : if true generate
        signal Sig0 : bit;
        signal Sig00 : bit;
    begin

    end generate g_1;

end architecture rtl;
```

jeremiah-c-leary, 2022-04-27:
Hey @hcommin,

I think I know what the issue is.  Now I am trying to figure out the best way to address it.

--Jeremy

jeremiah-c-leary, 2022-04-28:
Hey @hcommin,

I just pushed an update for this issue.  When you get a chance could you validate it works on your end.

I will plan to merge this to master by Apr 6th if I do not hear back from you.

Thanks,

--Jeremy

hcommin, 2022-04-28:
@jeremiah-c-leary A couple of my test cases are still failing. I don't get an unfixable error any more, but in some cases, the [generate_403](https://vhdl-style-guide.readthedocs.io/en/latest/generate_rules.html#generate-403) rule is not being applied. This seems to happen when there is yet another layer of nesting. For example:

```vhdl
entity e is
end entity e;

architecture rtl of e is
begin

    g_0 : if true generate
        g_1 : if true generate
            signal Sig0 : bit;
            signal Sig00 : bit;
        begin
        
            g_2 : if true generate

            end generate g_2;
            
        end generate g_1;
    end generate g_0;

end architecture rtl;
```

The colons are not being aligned:

```vhdl
      signal sig0 : bit;
      signal sig00 : bit;
```

If I delete the inner generate (`g_2`), then alignment is correct.

jeremiah-c-leary, 2022-04-30:
Hey @hcommin,

I pushed another update for this issue.  When you get a chance, could you validate it on your end.

Thanks,

--Jeremy

hcommin, 2022-05-01:
@jeremiah-c-leary One of my test cases is still failing, but now with an `IndexError: list index out of range`. The VHDL is very similar to above, except the innermost generate is a `for generate` instead of an `if generate`.

Here is a minimal example:
```vhdl
entity e is
end entity e;

architecture rtl of e is
begin

    g_0 : if true generate
        g_1 : if true generate
            signal Sig0 : bit;
            signal Sig00 : bit;
        begin
        
            g_2 : for i in 0 to 0 generate

            end generate g_2;
            
        end generate g_1;
    end generate g_0;
end architecture rtl;
```

Here is the full error message:
```
Traceback (most recent call last):
  File "C:\tools\vsg\vhdl-style-guide\bin\vsg", line 14, in <module>
    main()
  File "C:\tools\vsg\vhdl-style-guide\bin\..\vsg\__main__.py", line 129, in main
    fStatus, testCase, dJsonEntry, sOutputStd, sOutputErr = f((iIndex, sFileName))
  File "C:\tools\vsg\vhdl-style-guide\bin\..\vsg\apply_rules.py", line 108, in apply_rules
    oRules.check_rules(
  File "C:\tools\vsg\vhdl-style-guide\bin\..\vsg\rule_list.py", line 226, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "C:\tools\vsg\vhdl-style-guide\bin\..\vsg\rules\align_tokens_in_region_between_tokens_when_between_tokens_unless_between_tokens.py", line 62, in analyze
    lToi = oFile.get_tokens_bounded_by_token_when_between_tokens(self.left_token, self.right_token, self.lBetween[0], self.lBetween[1])
  File "C:\tools\vsg\vhdl-style-guide\bin\..\vsg\vhdlFile\vhdlFile.py", line 190, in get_tokens_bounded_by_token_when_between_tokens
    return extract.get_tokens_bounded_by_token_when_between_tokens(oLeft, oRight, oStart, oEnd, self.lAllObjects, self.oTokenMap, include_trailing_whitespace)
  File "C:\tools\vsg\vhdl-style-guide\bin\..\vsg\vhdlFile\extract\get_tokens_bounded_by_token_when_between_tokens.py", line 9, in get_tokens_bounded_by_token_when_between_tokens
    lMyPairs = get_token_pairs([oLeft, oRight], [oStart, oEnd], oTokenMap)
  File "C:\tools\vsg\vhdl-style-guide\bin\..\vsg\vhdlFile\extract\get_tokens_bounded_by_token_when_between_tokens.py", line 29, in get_token_pairs
    lLeft, lRight = oTokenMap.get_token_pair_indexes(lInnerPair[0], lInnerPair[1])
  File "C:\tools\vsg\vhdl-style-guide\bin\..\vsg\token_map.py", line 58, in get_token_pair_indexes
    return extract_start_end_indexes(lStartIndexes, lEndIndexes)
  File "C:\tools\vsg\vhdl-style-guide\bin\..\vsg\token_map.py", line 208, in extract_start_end_indexes
    lPairs = extract_pairs(mylStartIndexes, mylEndIndexes)
  File "C:\tools\vsg\vhdl-style-guide\bin\..\vsg\token_map.py", line 222, in extract_pairs
    lStartIndexes.remove(lPair[0])
IndexError: list index out of range
```

jeremiah-c-leary, 2022-05-01:
Hey @hcommin,

I had an issue with the function that was looking for token pairs.  If a pair was not found, then it would crash.  Could you check this latest fix on your code?

Thanks,

--Jeremy

hcommin, 2022-05-02:
@jeremiah-c-leary Thank you for your persistence with this - my test cases are now all passing.

PR #766 (2022-04-30): Updates for issue #762
Fixing issue with nested generates and rule generate_403.

  1)  Updated tests to expose the issue
  2)  Updated token_map.get_token_pair_indexes to honor nesting

Resolves #762
PR #769 (2022-05-02): Issue 762
Resolves #762
closed 2022-05-02 by hcommin