jeremiah-c-leary/vhdl-style-guide #763: port_map_003 behaving incorrectly

hcommin, 2022-04-27:
The behavior of [port_map_003](https://vhdl-style-guide.readthedocs.io/en/latest/port_map_rules.html#port-map-003) doesn't seem to match the documented behavior.

Here is a minimal example:
```vhdl
architecture rtl of e is
begin

  i_fifo : component fifo
    port map
    (
    WR_EN    => WR_EN,
    RD_EN    => RD_EN,
    OVERFLOW => OVERFLOW
    );

end architecture rtl;
```

The desired (and documented) behavior is:
```vhdl
    port map(
    WR_EN    => WR_EN,
    RD_EN    => RD_EN,
    OVERFLOW => OVERFLOW
    );
```

However, VSG (with default config) is keeping an empty line:
```vhdl
    port map (

      wr_en    => WR_EN,
      rd_en    => RD_EN,
      overflow => OVERFLOW
    );
```

This empty line is not being inserted by any other rule. If I remove this empty line and re-run VSG, then it is not re-added.

jeremiah-c-leary, 2022-04-27:
Hey @hcommin,

I just pushed a fix for this issue to the issue-763 branch.  When you a chance could you validate it is working on your end.

I will schedule this to be merged to master by May 6th if I do not hear back from you.

Thanks,

--Jeremy

hcommin, 2022-04-28:
@jeremiah-c-leary How do I run VSG from a Git branch? I cloned the repo and ran setup.py, which reports that it installed to `<python3>\lib\site-packages\vsg-3.10.0.dev8-py3.9.egg`, but then I don't know how to use this.

`vsg --version` still seems to pick up the default installation (`<python3>\lib\site-packages\vsg`):
```
VHDL Style Guide (VSG) version: 3.10.0
Git commit SHA: c4b05f8
```

So it doesn't recognize the new rule:
```
ERROR: Rule port_map_200 referenced in configuration could not be found
```

hcommin, 2022-04-28:
@jeremiah-c-leary I'm not sure what I'm doing wrong, but I still see the old version:
```
$ bin/vsg --version
VHDL Style Guide (VSG) version: 3.10.0
Git commit SHA: c4b05f8
```
I am definitely looking at the correct Git branch:
```
$ git status
On branch issue-763
Your branch is up to date with 'origin/issue-763'.

nothing to commit, working tree clean
```

And `git log -1` shows commit 9ff269a8f3860dc34c6385e4aaaa22fcbc3ab385, as expected. Any idea what I could try next?

jeremiah-c-leary, 2022-04-28:
@hcommin, have you tried uninstalling vsg via pip?

hcommin, 2022-04-28:
@jeremiah-c-leary Uninstalling vsg via pip seemed to work:

```
$ bin/vsg --version
VHDL Style Guide (VSG) version: 3.10.0.dev8
Git commit SHA: g9ff269a8
```
(But I'm not sure why there is a `g` preprended to the SHA hash). 

However, when I try to run VSG on any .vhd file, I now get a constant stream of errors:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Programs\Python3\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Programs\Python3\lib\multiprocessing\spawn.py", line 125, in _main
    prepare(preparation_data)
  File "C:\Programs\Python3\lib\multiprocessing\spawn.py", line 236, in prepare
    _fixup_main_from_path(data['init_main_from_path'])
  File "C:\Programs\Python3\lib\multiprocessing\spawn.py", line 287, in _fixup_main_from_path
    main_content = runpy.run_path(main_path,
  File "C:\Programs\Python3\lib\runpy.py", line 268, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Programs\Python3\lib\runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Programs\Python3\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\tools\vsg\vhdl-style-guide\bin\vsg", line 14, in <module>
    main()
  File "C:\Programs\Python3\lib\site-packages\vsg-3.10.0.dev8-py3.9.egg\vsg\__main__.py", line 137, in main
    with multiprocessing.Pool(commandLineArguments.jobs) as pool:
  File "C:\Programs\Python3\lib\multiprocessing\context.py", line 119, in Pool
    return Pool(processes, initializer, initargs, maxtasksperchild,
  File "C:\Programs\Python3\lib\multiprocessing\pool.py", line 212, in __init__
    self._repopulate_pool()
  File "C:\Programs\Python3\lib\multiprocessing\pool.py", line 303, in _repopulate_pool
    return self._repopulate_pool_static(self._ctx, self.Process,
  File "C:\Programs\Python3\lib\multiprocessing\pool.py", line 326, in _repopulate_pool_static
    w.start()
  File "C:\Programs\Python3\lib\multiprocessing\process.py", line 121, in start
    self._popen = self._Popen(self)
  File "C:\Programs\Python3\lib\multiprocessing\context.py", line 327, in _Popen
    return Popen(process_obj)
  File "C:\Programs\Python3\lib\multiprocessing\popen_spawn_win32.py", line 45, in __init__
    prep_data = spawn.get_preparation_data(process_obj._name)
  File "C:\Programs\Python3\lib\multiprocessing\spawn.py", line 154, in get_preparation_data
    _check_not_importing_main()
  File "C:\Programs\Python3\lib\multiprocessing\spawn.py", line 134, in _check_not_importing_main
    raise RuntimeError('''
RuntimeError:
        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.

        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:

            if __name__ == '__main__':
                freeze_support()
                ...

        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.
```
I'm looking into it, but please let me know if this error looks familiar to you.

jeremiah-c-leary, 2022-04-28:
That is bizarre...

It looks like something with the multiprocessing module.

I would try the `-p 1` command line option:
```mono
vsg -f <filename> --fix -p 1
```
which bypasses the multiprocessing module.

hcommin, 2022-04-28:
@jeremiah-c-leary Yeah, I was just trying `-p 1` and it worked!

Thanks for your help.

jeremiah-c-leary, 2022-04-28:
@hcommin,

Sweet...I do not know why you have to uninstall vsg though.  There must be something simple we are missing.

hcommin, 2022-04-28:
@jeremiah-c-leary I'm totally puzzled by it. However, my Python installations are a bit shaky and I've had various intermittent issues, like package dependencies getting mysteriously broken. So, most likely, the problem is on my side.

I can now confirm that enabling port_map_200 fixes the issue in my test cases. Closing the issue.

PR #764 (2022-04-28): Updates for issue #763
Adding rule to remove blank lines in a port map aspect.

  1)  Added documentation
  2)  Added tests
  3)  Added rule port_map_200

Resolves #763
closed 2022-04-28 by hcommin