jeremiah-c-leary/vhdl-style-guide #768: Document VSCode Integration

Zalfrin, 2022-04-29:
I was able to get vsg to run as a task in vscode and integrate nicely the problem reporting window. Parsing the output took some ugly regexps, but it works. Would it make sense to update the readthedocs Editor Integration page with this info?

Details on adding vscode integration:

Follow the instructions on https://code.visualstudio.com/docs/editor/tasks#_custom-tasks . This will create a user-level tasks.json file. You could also create this file in a project-level tasks.json file under the .vscode directory if you prefer.

In your tasks.json file, paste the following:

```
{
  // See https://go.microsoft.com/fwlink/?LinkId=733558
  // for the documentation about the tasks.json format
  "version": "2.0.0",
  "tasks": [
    {
      "label": "vsg",
      "type": "shell",
      "command": "vsg -f ${relativeFile}",
      //"command": "vsg -f ${relativeFile} -c vsg.json",
      "problemMatcher": {
        "owner": "vsg",
        "fileLocation": ["relative", "${workspaceFolder}"],
        "pattern": [
          {
            "regexp": "^File:  ([^\\s]*)\\s*$",
            "file": 1
          },
          { // match =======
            "regexp": "^.*$"
          },
          { // match Phase
            "regexp": "^.*$"
          },
          { // match Total Rules Checked
            "regexp": "^.*$"
          },
          { // match Total Violations
            "regexp": "^.*$"
          },
          { // match Error count
            "regexp": "^.*$"
          },
          { // match Warning count
            "regexp": "^.*$"
          },
          { // match ------
            "regexp": "^.*$"
          },
          { // match Rule/severity/line/solution header
            "regexp": "^.*$"
          },
          { // match ------
            "regexp": "^.*$"
          },
          {
            "regexp": "^  ([^\\s]*_\\d*)\\s*. ([^\\s]*)\\s*.\\s*([^\\s]*) . (.*)$",
            "code": 1,
            "severity": 2,
            "location": 3,
            "message": 4,
            "loop": true
          }
        ]
      }
    },
    {
      "label": "vsg-fix",
      "type": "shell",
      "command": "vsg -f ${relativeFile} --fix",
      "problemMatcher": {
        "owner": "vsg",
        "fileLocation": ["relative", "${workspaceFolder}"],
        "pattern": [
          {
            "regexp": "^File:  ([^\\s]*)\\s*$",
            "file": 1
          },
          { // match =======
            "regexp": "^.*$"
          },
          { // match Phase
            "regexp": "^.*$"
          },
          { // match Total Rules Checked
            "regexp": "^.*$"
          },
          { // match Total Violations
            "regexp": "^.*$"
          },
          { // match Error count
            "regexp": "^.*$"
          },
          { // match Warning count
            "regexp": "^.*$"
          },
          { // match ------
            "regexp": "^.*$"
          },
          { // match Rule/severity/line/solution header
            "regexp": "^.*$"
          },
          { // match ------
            "regexp": "^.*$"
          },
          {
            "regexp": "^  ([^\\s]*_\\d*)\\s*. ([^\\s]*)\\s*.\\s*([^\\s]*) . (.*)$",
            "code": 1,
            "severity": 2,
            "location": 3,
            "message": 4,
            "loop": true
          }
        ]
      }
    }
  ]
}
```

Now you can use CTRL+SHIFT+P to select "Tasks: Run Task". 

The "vsg" task will run vsg and add any issues to the vscode "Problems" window. 

The "vsg-fix" task will reformat the file using the --fix switch of vsg.

If you want any other vsg switches to be used, update the "command" key in the json as desired.

jeremiah-c-leary, 2022-04-30:
Hey @Zalfrin,

This is amazing.  I have been [asked about this before](https://github.com/jeremiah-c-leary/vhdl-style-guide/discussions/531), but never had an answer.

Thank you very much for figuring this out.

> Would it make sense to update the readthedocs Editor Integration page with this info?

I will definitely document this as it will help a lot of people.  I would also think we could find some way to commit the JSON file to the repo and point people to it.  That could make it easier to consume.

### Question 1

I noticed you had this line commented out:
```json
      //"command": "vsg -f ${relativeFile} -c vsg.json",
```
How do you handle configurations?

### Question 2

In the VSCode documentation they have this example under the problem matcher section:

[image]

I was wondering if you considered using the `-of syntastic` option.  It produces output similar to that and might easier to parse.

```mono
$ vsg -f original.vhd -of syntastic
ERROR: original.vhd(5)port_map_003 -- Move the ( to the same line as the *port map* keywords.
```

Again, thank you for figuring this out.

Regards,

--Jeremy

jeremiah-c-leary, 2022-05-02:
Hey @Zalfrin,

I came up with this task.json file:
```json
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "label": "vsg-check",
            "type": "shell",
            "command": "vsg -f ${relativeFile} -of syntastic",
            "problemMatcher": {
                "owner": "vsg",
                "fileLocation": ["relative", "${workspaceFolder}"],
                "pattern": {
                    "regexp": "^(.*):\\s(.*)\\(([0-9]+)\\)(.*)\\s--\\s(.*)\\s*$",
                    "severity": 1,
                    "file": 2,
                    "location": 3,
                    "code": 4,
                    "message": 5,
                    "loop": true
                },
            }
        },
        {
            "label": "vsg-fix",
            "type": "shell",
            "command": "vsg -f ${relativeFile} --fix -of syntastic",
            "problemMatcher": {
                "owner": "vsg",
                "fileLocation": ["relative", "${workspaceFolder}"],
                "pattern": {
                    "regexp": "^(.*):\\s(.*)\\(([0-9]+)\\)(.*)\\s--\\s(.*)\\s*$",
                    "severity": 1,
                    "file": 2,
                    "location": 3,
                    "code": 4,
                    "message": 5,
                    "loop": true
                },
            }
        }

    ]
}
```
Unfortunately, I found an error in my syntastic output.  It only outputs the last error.  I will be putting a fix in for it.

However, I did create this keybindings.json file:
```json
[
    {"key": "ctrl+f", "command": "workbench.action.tasks.runTask", "args": "vsg-fix"},
    {"key": "ctrl+i", "command": "workbench.action.tasks.runTask", "args": "vsg-check"}
]
```
Which is awesome, because it makes it trivial to run VSG within VSCode.

Still not sure the best way to handle configurations though.  I suppose you just put it in the command.  However, I wish there was an easier way to reference them.

You can  reference environment variables, and that may be a method someone could use to reference other commands.

[image]

Regards,

--Jeremy

jeremiah-c-leary, 2022-05-04:
Hey @Zalfrin,

I pushed the documentation updates to the issue-768 branch.  If you have a chance could you take a look and let me know if there are any changes I should make.

I will plan to merge this to master on April 6th if I do not hear back from you.

Thanks,

--Jeremy

Zalfrin, 2022-05-09:
@jeremiah-c-leary Looks good! My only feedback would be a screenshot of the problems window of vscode could be added, which eliminates the need to refer back to the terminal output and allows you to jump to the relevant line of code with a single click.

Regarding your questions, for configuration I would just include that in the task command. If you are working with different projects that use different configurations, you could uses a project-level tasks.json instead of a user-level one.

I did consider parsing the syntastic output instead of the standard output since it seemed like it be less verbose and easier to parse. I don't remember why I ended up going back to the standard output - perhaps it was because of the bug you mentioned, or maybe I just hadn't figured out how to use the "loop" property of the problem matcher yet. Actually, it may have been that I was having problems getting the multi-line problem matcher to work (https://code.visualstudio.com/docs/editor/tasks#_defining-a-multiline-problem-matcher). In any case, once I got something to work I didn't investigate any further.

Thanks for the keybinding tip! That's handy.

One feature that would be nice that I haven't investigated is having vsg run automatically upon file save. I think you could do something like this, but I don't have gulp installed: https://code.visualstudio.com/docs/editor/tasks#_background-watching-tasks

jeremiah-c-leary, 2022-05-12:
Hey @Zalfrin,

I added a small blurb about the problem window to the VSCode documentation.

> One feature that would be nice that I haven't investigated is having vsg run automatically upon file save. I think you could do something like this, but I don't have gulp installed: https://code.visualstudio.com/docs/editor/tasks#_background-watching-tasks

That would be even better than the keybinding because it would always run.

Thanks again for the figuring out the VSCode stuff.

--Jeremy