jeremiah-c-leary/vhdl-style-guide #773: Supporting multiple indentation configurations

hcommin, 2022-05-02:
I need to allow two different indentation configurations for `case` statements (simultaneously, in the same VHDL file). In other words, the code should not be fixed if it conforms to **either** configuration. If it conforms to **neither**, then it needs to be fixed, but I don't mind which style is implemented in the fix.

"Style 0" matches VSG's defaults. "Style 1" requires `when` to be aligned with `case`:

```
        case_statement:
            case_label:
                token : current
                after : current
            case_keyword:
                token : current
                after : '+1'
            end_keyword:
                token : '-1'
                after : '-1'
```

VHDL examples:

```vhdl
-- Style 0: "when" keyword is indented
case x is
    when 0 =>
        null;
    when others =>
        null;
end case

-- Style 1: "when" is aligned with "case"
case x is
when 0 =>
    null;
when others =>
    null;
end case
```

As far as I can see, this is not going to be possible with VSG. Many rules rely on the indentation configuration and it looks like I'm asking to break a fairly fundamental mechanism of VSG.

However, I thought I would just check. Is there any way to make this work? I'm struggling to find a workaround.

jeremiah-c-leary, 2022-05-04:
Hey @hcommin,

> As far as I can see, this is not going to be possible with VSG. Many rules rely on the indentation configuration and it looks like I'm asking to break a fairly fundamental mechanism of VSG.

Yeah, VSG is currently not setup to handle this case.  The indenting rules work on a token basis and this would require knowing how all the tokens are indented to then make a decision on what the correct indent should be.

I can not see a viable path forward in the current VSG implementation.

Regards,

--Jeremy

hcommin, 2022-05-04:
@jeremiah-c-leary Thanks - that makes sense.

closed 2022-05-04 by hcommin