jeremiah-c-leary/vhdl-style-guide #775: possibility to configure whitespace rules

Bonusbartus, 2022-05-03:
Currently it is possible to enable or disable a lot of whitespace rules, but in some cases some more configuration would be nice.
for instance:
I prefer to have a whitespace before and after an opening parenthesis, unless the next symbol is also an opening (or closing) parenthesis:
```vhdl
if rising_edge ( clk ) then
```
or
```vhdl
 if (( A = '1' ) and ( B = '1' )) then
```

also maybe an option for each whitespace rule to select between allowing 
no whitespaces
one whitespace
at least one whitespace

this last one I tend to use to align if and elsif statements:

```vhdl
if    (( A = '1' ) and ( B = '1' )) then
  null;
elsif (( A = '1' ) and ( B = '0' )) then
  null;
end if;
```

jeremiah-c-leary, 2022-05-03:
Hey @Bonusbartus,

> also maybe an option for each whitespace rule to select between allowing
no whitespaces
one whitespace
at least one whitespace

I think that is a great idea.  I currently have several base rules that enforce a different number of spaces, but there is no reason I could not collapse them all into a single whitespace rule with several options:
```yaml
  number_of_whitespaces: 1
  number_of_whitespaces: 4
  number_of_whitespaces: 0
  number_of_whitespaces: '>1'
  number_of_whitespaces: '>5'
```
I can default all the current rules to 1, but then you would have the ability to control the number of spaces.
> I prefer to have a whitespace before and after an opening parenthesis, unless the next symbol is also an opening (or closing) parenthesis:
Maybe we could add an `unless` option?
```yaml
  number_of_whitespaces: 1
  unless:  '('
  unless_whitespaces: 0
```
Although defining unless could be tricky depending on what the whitespace is between.
We will have to think a bit on that.

Does it sound like a reasonable implementation?

--Jeremy

Bonusbartus, 2022-05-03:
Hi @jeremiah-c-leary,

I think the number_of_whitespaces option would be perfect.

I am indeed not sure about the unless option. I think the only case where this is used (at least in my case) is with multiple opening or closing parentheses. maybe it would be better to make a dedicated whitespace rule for opening parenthesis and for closing parenthesis?
a rule: "enforce number_of_whitespaces after token" (where token can be opening or closing parenthesis) which can skip the whitespaces if current token and next token are the same (opening or closing parenthesis).

Thanks!

jeremiah-c-leary, 2022-05-03:
Hey @Bonusbartus,

> maybe it would be better to make a dedicated whitespace rule for opening parenthesis and for closing parenthesis?

That could work.  It would use the same whitespace logic, but with `number_of_whitespaces` set to 0.

--Jeremy

hcommin, 2022-05-04:
@jeremiah-c-leary Please let me know if I should create a separate issue, but I have a very closely related feature request.

Just being able to set a minimum like this would already be great:
```
number_of_whitespaces: '>5'
```

However, in some cases, it would be even better if the whitespace could optionally be rounded up to a multiple of the tab width **from the start of the line**. (I'm thinking mostly about colon alignment). In other words, at least 5 spaces are inserted, but if the number of characters from the start of the line is not a multiple of the tab width, then additional spaces are inserted.

I don't know if this would be possible, but it would be a very useful feature in my use cases.

jeremiah-c-leary, 2022-05-13:
Hey @hcommin,

I could add another option.  Maybe something like this:

```yaml
  number _of_whitespaces: '>5'
  align_to_next_tab: true
```
I believe we would use the `IndentSize` option for the rule to know the tab size, or we manually set a tab size.  Hmm...not sure of the best way to handle that yet.

--Jeremy

jeremiah-c-leary, 2022-05-14:
Hey @Bonusbartus and @hcommin,

I pushed an initial implementation for this issue to the issue-775 branch.

I was able to use this configuration:
```yaml
rule:
    if_003:
        number_of_spaces: 4
    if_005:
        number_of_spaces: 1
```
and was able to recreate the spacing from the example above:
```vhdl
if    (( A = '1' ) and ( B = '1' )) then
  null;
elsif (( A = '1' ) and ( B = '0' )) then
  null;
end if;
```
When you get a chance could you give it a try and let me know what you think.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-05-23:
Hey @Bonusbartus,

Just wanted to ping you on this issue to see if it addresses your issue.

Thanks,

--Jeremy

Bonusbartus, 2022-05-23:
Hi @jeremiah-c-leary,

Sorry, must have missed the previous update, I'll look into the issue775 branch implemention. Seems to be perfect from your description. 
I'll get back to you when I performed some tests.

Regards
Bart

Bonusbartus, 2022-05-30:
Hey @jeremiah-c-leary,

played around with this and seems to work nicely. Had some trouble figuring out which rules this would apply to, and which rules it doesn't.
Would it make sense to put this into a generic rule_group? the difference in having a default number of spaces == 1, or >=1 would fix a lot of errors ;)

also I found the ">" a little bit counter-intuitive -> I ended up setting the number_of_spaces: ">0"
would it make more sense to use something like "1+" there not being a simple ">=" sign on our keyboards?
not really a problem though!

regards

Bart

jeremiah-c-leary, 2022-05-30:
Hey @Bonusbartus,

> Would it make sense to put this into a generic rule_group?

There is already a `whitespace` rule group.  Try setting the spacing for all whitespace rules using:
```yaml
rule:
  group:
    whitespace:
        number_of_spaces: '2'
```
It will only apply to those rules which have that configuration option.

> also I found the ">" a little bit counter-intuitive -> I ended up setting the number_of_spaces: ">0"
would it make more sense to use something like "1+" there not being a simple ">=" sign on our keyboards?
not really a problem though!

I also found the `>` to be a bit confusing.  I do like your `1+` idea though, as that sets the minimum number and the plus indicates it could be more.

I will change `>` to a `+` and we can see how it works out.

--Jeremy

jeremiah-c-leary, 2022-05-30:
Hey @Bonusbartus,

I added the `+` option and updated the documentation.  Were you aware of a `>=` option in addition to the `>` option.

Let me know what you think of the `+` option.

--Jeremy

Bonusbartus, 2022-05-31:
Hey @jeremiah-c-leary 
that solution looks perfect. No I was not aware yet. the + sign is perfect for me.

Bart

jeremiah-c-leary, 2022-06-01:
Hey @Bonusbartus,

That is awesome.  I will plan to merge this to master on 06/03/2022 unless you have any other updates before then.

Thanks,

--Jeremy

PR #782 (2022-06-03): Updates for Issue 775
Resolves #775
closed 2022-06-03 by jeremiah-c-leary