jeremiah-c-leary/vhdl-style-guide #776: Missing Alignment option for USE clause

Bonusbartus, 2022-05-04:
Hi @jeremiah-c-leary 

it seems like the use clause is missing from the alignment options:
```vhdl
"use_clause": {
        "keyword": {
          "after": "current",
          "token": "current"
        }
      },

```
seems to not work. am I doing something wrong or is it indeed missing?

Bart

jeremiah-c-leary, 2022-05-04:
Hey @Bonusbartus,

What code change are you trying to enforce?

--Jeremy

Bonusbartus, 2022-05-04:
Hi @jeremiah-c-leary 

code like this:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
```

Bart

jeremiah-c-leary, 2022-05-04:
Hey @Bonusbartus,

Try adding this to a configuration:
```yaml
rule:
    library_008:
        indentSize : 0
```

--Jeremy

Bonusbartus, 2022-05-04:
Thanks @jeremiah-c-leary ! Works perfectly!
I didn't know this configuration argument was possible.
is it documented somewhere? How and for which rules does it work?
what has precedence in this case, the indent part of the configuration or the rule: indentSize?

regards,

Bart

jeremiah-c-leary, 2022-05-04:
Hey @Bonusbartus,

You can check all the rule options using the `-rc` command.
```mono
$ vsg -rc library_008
{
  "rule": {
    "library_008": {
      "indentSize": 2,
      "phase": 4,
      "disable": false,
      "fixable": true,
      "severity": "Error"
    }
  }
}
```
> is it documented somewhere?

Check the [configuration overview](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_overview.html#rule).
There are also a couple of examples of setting the indent size.
Not sure if that would have been enough for you to figure this out on your own though.

> How and for which rules does it work?

That parameter is available on all rules, but is only used on indent rules.

> what has precedence in this case, the indent part of the configuration or the rule: indentSize?

The indent rules take the indent level based on the configuration and multiply it by the indentSize of the rule.  So in your case the indent level is 1, but we set the indentSize to 0 so it left aligns it.

--Jeremy

Bonusbartus, 2022-05-04:
Ah yes, I understand.
I did read that part of the configuration overview, but didn't think of
using the size to actually change the indent level back to zero.

Thanks!

jeremiah-c-leary, 2022-05-05:
Hey @Bonusbartus,

Are we good to close this issue?

--Jeremy

Bonusbartus, 2022-05-06:
Yes, this solution works for me.

closed 2022-05-06 by Bonusbartus