jeremiah-c-leary/vhdl-style-guide #777: Add a rule to enforce case of predefined attributes

Bonusbartus, 2022-05-04:
Hi,

I think there is currently no rule to enforce the case for predefined attributes:
```vhdl

Type attributes:
T'ASCENDING
T'BASE 
T'HIGH
T'IMAGE(X)
T'LEFT
T'LEFTOF(X)
T'LOW
T'POS(X)
T'PRED(X)
T'RIGHT
T'RIGHTOF(X)
T'SUCC(X)
T'VAL(X)
T'VALUE(X)

Array attributes
A'ASCENDING[(N)]
A'HIGH[(N)]
A'LEFT[(N)]
A'LENGTH[(N)]
A'LOW[(N)]
A'RANGE[(N)]
A'REVERSE_RANGE[(N)]
A'RIGHT[(N)]

S'ACTIVE
S'DELAYED[(T)]
S'DRIVING
S'DRIVING_VALUE
S'EVENT
S'LAST_EVENT
S'LAST_ACTIVE
S'LAST_VALUE
S'QUIET[(T)]
S'STABLE[(T)]
S'TRANSACTION

Everything attributes
E'INSTANCE_NAME
E'PATH_NAME
E'SIMPLE_NAME

```

it would be nice to have a rule for this similar to all the other case rules.

thanks

sorry for all the requests ;) I'd be glad to help implement or test them

jeremiah-c-leary, 2022-05-04:
Hey @Bonusbartus,

I can add a rule to enforce the case for those.

--Jeremy

jeremiah-c-leary, 2022-05-05:
Hey @Bonusbartus,

I just pushed an update to the issue-777 branch.  When you get a chance could you check it out and let me know what you think.

I will plan to merge this to master on May 13th if I do not hear back from you.

Thanks,

--Jeremy

Bonusbartus, 2022-05-08:
Hi @jeremiah-c-leary,

Thanks for the addition. Seems to be working fine as far as I can tell!

jeremiah-c-leary, 2022-05-08:
Awesome, I will merge this to master.

--Jeremy

PR #779 (2022-05-08): Updates for issue #777
Adding rule to enforce case of predefined attributes.

  1)  Added documentation
  2)  Added test
  3)  Added rule attribute_500

Resolves #777
closed 2022-05-08 by jeremiah-c-leary