jeremiah-c-leary/vhdl-style-guide #780: Add rule to check indentation for 'else generate'

staerz, 2022-05-12:
**Is your feature request related to a problem? Please describe.**
There's a gap in the matrix that was discovered in our code base: There is no VSG rule to check the indentation of `else generate`.

**Describe the solution you'd like**
It would be nice to have a rule added also for the indentation of `else generate`.

The only indentation checks done on a generate block are for the begin keyword (generate_006) and the end keyword (generate_007).
There should be a similar check for else.

In particular, VSG reports no errors on the following:

```vhdl
  gen_felix_clocks : if L_INJECTOR_B_FELIX_CLK generate
  begin
    felix_clk <= fmca_gbtclk_m2c_p(0);
    refclk    <= fmca_gbtclk_m2c_p(1);
                                                else generate
    refclk    <= refclk_fmca_p;
  end generate gen_felix_clocks;
```

**Describe alternatives you've considered**
There is no alternative :rofl:

**Additional context**
Although VSG is covering so many things it's always nice to still be able to find small things to even improve it more.

jeremiah-c-leary, 2022-05-13:
Hey @staerz,

> Although VSG is covering so many things it's always nice to still be able to find small things to even improve it more.

Looks like I was also missing indents for `elsif generate`.  There are also a slew of rules I could add to ensure proper formatting.  One of my future tasks is to scrub the LRM to search for missing rules.

I pushed an update to the issue-780 branch.  When you get a chance could you validate it on your end.

I will plan to merge it to master on May 20th if I do not hear from you.

Thanks,

--Jeremy

staerz, 2022-05-16:
Hi @jeremiah-c-leary,

I do confirm that now the `else` indentation is also checked and `if_generate_statement_301` would shout out.

From the commit that you did I do though not see how that's actually working and in particular I do not see documentation added for that rule, but I may be misreading the commit.

jeremiah-c-leary, 2022-05-16:
Hey @staerz,

> From the commit that you did I do though not see how that's actually working and in particular I do not see documentation added for that rule, but I may be misreading the commit.

You can check the [pull request](https://github.com/jeremiah-c-leary/vhdl-style-guide/pull/781/files) to see which files have changed.  You will see I added documentation for the new rules `generate_019`, `if_generate_statement_300`, and `if_generate_statement_301`.

I moved the rule documentation into the docstring of every rule.  I then created a test to compare the contents of the docstring against the restructured text documentation.  The tests will fail if they do not match.  This helps me ensure the generated documentation matches the actual rules.

--Jeremy

staerz, 2022-05-16:
Alright. Well, if you have had another look over that and feel confident then let's merge this :)

jeremiah-c-leary, 2022-05-16:
Sweet...merging now.

PR #781 (2022-05-16): Updates for issue #780:
Adding rule to enforce indenting of the `end` keyword:

  1)  Added documentation
  2)  Added test
  3)  Added rule generate_300

Resolves #780
closed 2022-05-16 by jeremiah-c-leary