jeremiah-c-leary/vhdl-style-guide #784: Align attribute specifications also at the colon

jobtijhuis-philips, 2022-05-16:
Currently attribute specifications are not aligned after the `of` keyword, so the following is not aligned:
```vhdl
  attribute mark_debug of send_count     : signal is "true";
  attribute mark_debug of sig_pos  : signal is "true";
  attribute mark_debug of write_index  : signal is "true";
  attribute mark_debug of init_txn_pulse  : signal is "true";
  attribute mark_debug of m_axi_wdata  : signal is "true";
```
But it would be more readable if attributes are also aligned at the colon `:`

Ideally the previous snippet would look like the following after running the formatter:
```vhdl
  attribute mark_debug of send_count     : signal is "true";
  attribute mark_debug of sig_pos        : signal is "true";
  attribute mark_debug of write_index    : signal is "true";
  attribute mark_debug of init_txn_pulse : signal is "true";
  attribute mark_debug of m_axi_wdata    : signal is "true";
```

**As a note:** The current rules do correctly not touch the spaces after the `of` keyword, so you can fix it manually.

jeremiah-c-leary, 2022-05-16:
Hey @jobtijhuis-philips,

> As a note: The current rules do correctly not touch the spaces after the of keyword, so you can fix it manually.

Could you elaborate on this?  Rule attribute_specification__001 which enforces a single space after the `of` keyword.

--Jeremy

jobtijhuis-philips, 2022-05-16:
> Hey @jobtijhuis-philips,
> 
> > As a note: The current rules do correctly not touch the spaces after the of keyword, so you can fix it manually.
> 
> Could you elaborate on this? Rule attribute_specification__001 which enforces a single space after the `of` keyword.
> 
> --Jeremy

I just meant that there was nothing wrong with the current implementation and that this is just a pure enhancement.

jobtijhuis-philips, 2022-05-16:
I just checked it out. That is exactly what I meant!

Great work on the quick fixing!

jeremiah-c-leary, 2022-05-16:
Hey @jobtijhuis-philips,

> I just checked it out. That is exactly what I meant!

Awesome.  I will assume that means I can merge this to master and close this issue.

--Jeremy

PR #785 (2022-05-16): Updates for issue #784:
Adding rules to align colons in attribute specifications:

  1)  Added documentation
  2)  Added tests
  3)  Added the following rules

      architecture_400
      block_402
      package_402
      process_401
      subprogram_body_401

Resolves #784
closed 2022-05-16 by jeremiah-c-leary