jeremiah-c-leary/vhdl-style-guide #789: variable_assignment_001

timkoers, 2022-07-04:
In our code, we use
```
(
  test.one,
  test.two,
  test.three
) := vec;
```
quite often, but this isn't supported by rule variable_assignment_001. 

That the rule looks at the context, ie are there parentheses, if yes, add another indent level/single space (<-- rule parameter).

jeremiah-c-leary, 2022-10-09:
Morning @timkoers ,

I pushed an update to the `issue-789` branch for this issue.  When you get a chance could you check it out on your end.

Regards,

--Jeremy

jeremiah-c-leary, 2022-10-18:
Hey @timkoers ,

Just wanted to ping you to see if you had a chance to checkout the latest update.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-11-07:
Morning @timkoers ,

Just wanted to ping you to see if you had a change to validate the update.

Thanks,

--Jeremy

timkoers, 2022-11-11:
Sorry!

I'll try to check it on monday!

jeremiah-c-leary, 2022-12-02:
Hey @timkoers ,

Just wanted to ping you to see if you had a change to validate the update.

--Jeremy

timkoers, 2022-12-17:
Not yet, I've added it to my agenda to do on the 2nd of january!

Sorry!

timkoers, 2023-01-03:
It's working! Thanks :-)

PR #841 (2024-03-09): Updates for issue #789
Added classification of aggregates for targets in simple_variable_assignments.

  1)  Added tests
  2)  Added classifiers
      a)  target
      b)  aggregate
  3)  Updated indent configuration

Resolves #789
closed 2023-01-03 by timkoers