jeremiah-c-leary/vhdl-style-guide #791: concurrent_003 and procedure_call_400

timkoers, 2022-07-04:
I can't align the concurrent statements and procedure calls like this:

```
begin
  (
    test        => '1',
    test_two => '2',
  )
```

or 

```
begin
  testproc
  (
    test        => '1',
    test_two => '2',
  )

I'd like to align to the parantesis, but with an extra indent

jeremiah-c-leary, 2022-07-04:
Hey @timkoers,

The rule `concurrent_003` would not be applied in your examples as they are not concurrent signal assignments.

As for formatting of the procedure call...if you ran into the following example:
```vhd
testproc(test =>
'1', test_two
=> '2')
```
How would you want it formatted?

timkoers, 2022-07-04:
> The rule concurrent_003 would not be applied in your examples as they are not concurrent signal assignments.
That is correct, however, the same structure is used for concurrent assignments.

The example above could be formatted in two ways, but since statement already contains a new line, I'd keep both sides of the operand (=>) on the same line

jeremiah-c-leary, 2022-07-04:
Hey @timkoers,

Could you post a version of my example into your preferred format?  It helps me understand what you are looking for and then I use the examples in my tests to ensure the formatting is correct.

--Jeremy

timkoers, 2022-07-09:
Sure!

```
testproc (
  test     =>'1', 
  test_two => '2'
)
```

And
```
(
  test,
  test_two,
) := test_three;
```

jeremiah-c-leary, 2022-07-16:
Hey @timkoers,

I have pushed an  update for this to the issue-791 branch.  Currently only `concurrent_procedure_call`s are being checked and fixed.  When you get a chance could you run the latest on the branch and let me know if there are any issues?

In my tests I used the following configuration options:
```yaml
rule:
  procedure_call_003:
    first_open_paren : 'remove_new_line'
    last_close_paren : 'add_new_line'
    association_list_comma : 'remove_new_line'
    association_element : 'add_new_line'
  procedure_call_400:
    align_left : 'yes'
    align_paren : 'no
```
 to match the following format:
```vhdl
testproc (
  test     => '1', 
  test_two => '2'
);
```

Thanks,

--Jeremy

jeremiah-c-leary, 2022-07-23:
Hey @timkoers,

Just wanted to ping you to see if you had a chance to check on the update I pushed for this issue.

Thanks,

--Jeremy

timkoers, 2022-07-26:
Life has been busy, I'll get back to you tomorrow!

timkoers, 2022-07-27:
It's working at I'd expect! 

Thanks, are there any issues you like me to try?

jeremiah-c-leary, 2022-07-27:
Hey @timkoers,

> It's working at I'd expect!

Sweet!!!

I will merge this to master so it can be included in the next release.

> Thanks, are there any issues you like me to try?

Could you elaborate?  I am assuming by issues you are referring to the tickets I have open.

--Jeremy

PR #799 (2022-07-29): Updates for issue #791
Adding rule procedure_call_401

  1)  Added documentation
  2)  Added test
  3)  Added rule

Resolves #791
closed 2022-07-29 by jeremiah-c-leary