jeremiah-c-leary/vhdl-style-guide #802: alignment rules not working on "when assignment" within process

oletf, 2022-07-13:
```vhdl
process (clk) is
begin
  a1 <= '0';
    b1 <= '0' when a1 = '1;
     a2 <= '0';
         b2 <= '0' when a1 = '1';
end process;
```

in my case here `a1` and `a2` become aligned, but `b1` and `b2` always stay where they are.
I couldn't determine which rule would be supposed to take care of this

jeremiah-c-leary, 2022-07-13:
Hey @oletf,

Just to be clear, are you saying VSG formats the code to this:
```vhdl
process (clk) is
begin
  a1 <= '0';
    b1 <= '0' when a1 = '1;
  a2 <= '0';
         b2 <= '0' when a1 = '1';
end process;
```

--Jeremy

jeremiah-c-leary, 2022-07-13:
Hey @oletf,

I just pushed a fix for this to the issue-802 branch.  If you could check it out and let me know if addresses the issue I would appreciate it.

I plan to merge this to master on July 22nd if I do not hear back from you.

Thanks,

--Jeremy

oletf, 2022-07-15:
issue is now fixed, thanks a lot!

PR #803 (2022-07-15): Updates for issue #802
Adding conditional_waveform_assigment to rule sequential_001.

  1)  Added tests
  2)  Updated indent rules
  3)  Updated rule sequential_001

Resolves #802
closed 2022-07-15 by oletf