jeremiah-c-leary/vhdl-style-guide #806: VHDL-2008 block comment with error

el-coder-sb, 2022-07-20:
**Environment**
Version 3.11 on Windows10

**Describe the bug**
When using keywords like `entity` inside block comment which was introduced in vhdl- 2008 I get `Error: Unexpected token detected [...]`

**To Reproduce**
Steps to reproduce the behavior:

1. Run: `vsg -f main.vhd` with the following file content:

```vhdl
/* This entity does something */

entity main is
end entity main;

architecture arch of main is
begin
end architecture arch;
```

2. See error:
```
Error: Unexpected token detected while parsing entity_declaration @ Line 2, Column 18 in file main_3r.vhd
       Expecting : is
       Found     : something
```

**Expected behavior**
No error.

jeremiah-c-leary, 2022-07-20:
Hey @el-coder-sb,

Well, I learned something new today.

All of my parsing, classifications, and rules are geared towards `single-line` comments.  We may need different rules to handle `delimited comments`.

I will start with just being able to parse and classify them.

Regards,

--Jeremy

el-coder-sb, 2022-07-21:
> Well, I learned something new today.

It was a good day for you :-)

> I will start with just being able to parse and classify them.

Sounds good! That would be enough for the moment. Just let me know if you need further information/support.

jeremiah-c-leary, 2022-07-23:
Hey @el-coder-sb,

I just pushed an update to the `issue-806` branch.  When you get a chance could you check it out and let me know how it goes?

Thanks,

--Jeremy

el-coder-sb, 2022-07-27:
Hey @jeremiah-c-leary,

looks good. I just formatted successfully my file where the error ocurred.

Thanks for the quick patch!

jeremiah-c-leary, 2022-07-27:
Hey @el-coder-sb,

Awesome!!.  I will merge to the master so it can be in the next release.

--Jeremy

PR #808 (2022-07-27): Issue 806
Resolves #806
closed 2022-07-27 by jeremiah-c-leary