jeremiah-c-leary/vhdl-style-guide #807: Unintendet alignment of comments

el-coder-sb, 2022-07-22:
I´m not sure why the app does what it does. Which parameter does determine this:
Inside a case statement in-line comments are aligned through the whole case block.

I couldn´t find any parameter to avoid this. 

cmd: `vsg -f test_comments.vhd --fix`

before:

```vhdl
            case main_rsu_ctrl_fsm_st_s is
                when GET_ANF_ST =>
                        if CONN_TO_rsu_inst_rsu_busy_sl_o = '0' then -- 2 Takte
                            CONN_TO_rsu_inst_rsu_param_slv_i <= "101"; -- aha
                            CONN_TO_rsu_inst_rsu_read_param_sl_i <= '1'; -- bla
                        end if;

                when SET_WDG_VALUE_ST =>
                    if ffs_busy_slv_s(0) = '1' and CONN_TO_rsu_inst_rsu_busy_sl_o = '0' then   -- falling edge
                       main_rsu_ctrl_fsm_st_s <= SET_ADDR_ST;
                    end if;

            end case;
```

After running vsg:

```vhdl

            case main_rsu_ctrl_fsm_st_s is
                when GET_ANF_ST =>
                    if (CONN_TO_rsu_inst_rsu_busy_sl_o = '0') then                               -- 2 Takte
                        CONN_TO_rsu_inst_rsu_param_slv_i     <= "101";                           -- aha
                        CONN_TO_rsu_inst_rsu_read_param_sl_i <= '1';                             -- bla
                    end if;

                when SET_WDG_VALUE_ST =>
                    if (ffs_busy_slv_s(0) = '1' and CONN_TO_rsu_inst_rsu_busy_sl_o = '0') then   -- falling edge
                        main_rsu_ctrl_fsm_st_s <= SET_ADDR_ST;
                    end if;

            end case;
```

Full code sample:

```vhdl
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

entity remote_update_top is
end entity remote_update_top;

architecture remote_update_top_arc of remote_update_top is

begin
    rsu_ctrl_fsm: process(clk_rsu_sl_s, nRESET_SL_I)
    begin
        if nRESET_SL_I = '0' then
            anf_sl_s                <= '0';
        elsif rising_edge(clk_rsu_sl_s) then
            case main_rsu_ctrl_fsm_st_s is
                when GET_ANF_ST =>
                        if CONN_TO_rsu_inst_rsu_busy_sl_o = '0' then -- 2 Takte
                            CONN_TO_rsu_inst_rsu_param_slv_i <= "101"; -- aha
                            CONN_TO_rsu_inst_rsu_read_param_sl_i <= '1'; -- bla
                        end if;

                when SET_WDG_VALUE_ST =>
                    if ffs_busy_slv_s(0) = '1' and CONN_TO_rsu_inst_rsu_busy_sl_o = '0' then   -- falling edge
                       main_rsu_ctrl_fsm_st_s <= SET_ADDR_ST;
                    end if;

            end case;
        end if;
    end process rsu_ctrl_fsm;
end architecture remote_update_top_arc; -- of remote_update_top
```

jeremiah-c-leary, 2022-07-22:
Hey @el-coder-sb,

Alignment rules cover multiple lines and are therefore covered by a rule in the enclosing hierarchy.  So in your example the hierarchy is:
```mono
architecture
  process
    if
      case
```
If you scan the `case` rules for those dealing with `alignments`, you will only find `case_011` and `case_019`.  Neither rule match the behavior you are looking for.

Going up the hierarchy, you would then look at the `if` rules and find `if_009`, `if_032`, and `if_033`.  Again, neither rule match the behavior.

Going another level up the hierarchy you would look at the `process` rules.  Here you will find `process_020`, `process_028`, `process_031`, `process_033`, `process_034`, `process_035`, `process_400`, and `process_401`.  Of these rules it looks like `process_035` is the rule enforcing the alignment of comments.

I hope that explanation helps.

Regards,

--Jeremy

el-coder-sb, 2022-07-27:
@jeremiah-c-leary yepp, `process_035` is correct and the explanation right helpfull. Thanks!

closed 2022-07-27 by el-coder-sb