jeremiah-c-leary/vhdl-style-guide #813: VHDL equality/empty configuration not detected => unexpected token found

simoneruffini, 2022-07-24:
**Environment**
NixOS from flake, vsg version 3.11.0

**Describe the bug**
In a (correctly written) VHDL file with an empty configuration, vsg returns the following:
```
Error: Unexpected token detected while parsing configuration_declaration @ Line 66, Column 4 in file src/acc/acc.vhd
       Expecting : end
       Found     : FOR
```
This is the VHDL file
```vhdl
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.std_logic_unsigned.all;

entity ACC is
  generic (
    NBIT : integer := 32
  );
  port (
    A           : in    std_logic_vector(NBIT - 1 downto 0);
    B           : in    std_logic_vector(NBIT - 1 downto 0);
    CLK         : in    std_logic;
    RST_N       : in    std_logic;
    ACCUMULATE  : in    std_logic;
    Y           : out   std_logic_vector(NBIT - 1 downto 0)
  );
end entity ACC;

architecture BEHAVIORAL of ACC is

  signal tmp : std_logic_vector(NBIT - 1 downto 0);

begin

  ALU_P : process (CLK, RST_N) is
  begin

    if (RST_N = '0') then
      tmp <= (OTHERS => '0');
    elsif (CLK'event and CLK = '1') then
      if (ACCUMULATE = '1') then
        tmp <= tmp + A;
      else
        tmp <= A + B;
      end if;
    end if;

  end process ALU_P;

  Y <= tmp;

end architecture BEHAVIORAL;

CONFIGURATION CFG_ACC_BEHAVIORAL OF ACC IS
   FOR BEHAVIORAL 
   END FOR BEHAVIORAL;
END CFG_ACC_BEHAVIORAL;
```
If the `FOR BEHAVIORAL END FOR BEHAVIORAL;` lines are commented out vsg works correctly but the configuration has no meaning.

**To Reproduce**
run the vhdl file with `-ap -p 6 --style jcl` and the followig rules in a json file 
```json
{
  "rule": {
    "constant_004": {
      "case": "upper"
    },
    "constant_015": {
      "disable": false,
      "fixable": true
    }
  }
}
```

**Expected behavior**
Vsg should find the internal for and not an unexpected token. Check this https://vhdlwhiz.com/configuration-constructs-explained/

jeremiah-c-leary, 2022-07-26:
Hey @simoneruffini,

This is a known limitation of VSG at the moment.  I did not get a chance to finish my configuration classifier.  This is more a medium term task for me at the moment.  If possible, I would move the configuration to another file until I can get to this.

Regards,

--Jeremy

simoneruffini, 2022-07-26:
Unfortunately I can't move the declarations to another file.
Is there a way to disable the syntax checking of vsg in certain parts?

jeremiah-c-leary, 2022-07-26:
Hey @simoneruffini,

> Is there a way to disable the syntax checking of vsg in certain parts?

Unfortunately not.

> Unfortunately I can't move the declarations to another file.

I will push this issue to the top of list.

--Jeremy

jeremiah-c-leary, 2022-08-03:
Hey @simoneruffini,

I was running my classifier updates against your example and ran into a little issue.  The following snippet:
```vhdl
FOR BEHAVIORAL 
END FOR BEHAVIORAL;
```
is a `block_configuration`.  However, the LRM defines it as:
```mono
block_configuration ::=
    for block_specification
        { use_clause }
        { configuration_item }
    end for ;
```
The last `BEHAVIORAL` in your example is not allowed per the LRM.

I am curious which compiler you are using.

--Jeremy

jeremiah-c-leary, 2022-08-03:
Hey @simoneruffini,

I just pushed an update for this issue to the `issue-813` branch.  When you get a chance could you check it out and let me know if it is working for you.

I will plan to merge this to master by August 12th if I do not hear back from you.

Thanks,

--Jeremy

simoneruffini, 2022-08-03:
Hi I used vsim 10.7c  from questasim (simens) suite. Moreover if I remember correctly ghdl supports it too.

For what concerns the update I will try it shortly once I figure out how to compile your code under NixOS.

jeremiah-c-leary, 2022-08-03:
Hey @simoneruffini,

Before you check out the update, let me add in the parsing of the last `BEHAVIORAL`, otherwise your files will fail to parse.

--Jeremy

jeremiah-c-leary, 2022-08-04:
Hey @simoneruffini,

I just pushed an update that should work for you.  Let me know how it works for you.

Thanks,

--Jeremy

PR #814 (2022-08-13): Updates for issue #813
Adding ability to classify configurations:

  1) Added additional tokens
  2) Added classifiers
  3) Updated test

Resolves #813
closed 2022-08-13 by jeremiah-c-leary